acmerobotics/road-runner-quickstart #129: Robot is Running Way Farther than its Trajectory

MischeviousMasterMind, 2021-02-21:
When I ran the straight test, I noticed that the robot is running way longer than it is supposed to. I checked the dashboard and saw that according to the graph, the robot is currently running a lot slower than it actually is. To investigate further, I set the distance of the Straight Test to one inch and then ran the code. What I saw, it seemed that the robot was running one tile instead of one inch. And on the graph, it seemed that it is actually running one inch. So, I multiplied my gear ratio by 24 and then tried again. What I saw was that the robot was still running one tile, but on the dashboard, it showed the robot running one tile and the end pose to be one inch. It is seems that the robot kept on overshooting. 

To prove this further, I set the distance of the Straight Test to 4 inches and the MAX_VEL and MAX_ACCEL variables to 2 and it was repeating the same behavior as my last test. Instead of 4 inches, the robot ran 4 tiles and the endpose was still 4 inches. The robot doesn't notice that it is overshooting and still runs instead of decelerating. 

[The Link to the Video](https://user-images.githubusercontent.com/73049474/108640290-a8e7cf00-744d-11eb-9e73-5490a9cb6c90.mp4
)

I'm using the 5202 Series Yellow Jacket Planetary Gear Motor [(5202-0002-0019)](https://www.gobilda.com/5202-series-yellow-jacket-planetary-gear-motor-19-2-1-ratio-312-rpm-3-3-5v-encoder/) and the 96mm Mecanum Wheel Set [(3213-3606-0002)](https://www.gobilda.com/96mm-mecanum-wheel-set-70a-durometer-bearing-supported-rollers/). I am using drive encoders. This is what my Drive Constants looked like before I multiplied the gear ratio by 24 and set the MAX_VEL and MAX_ACCEL to 2:

    
    public static final double TICKS_PER_REV = 537.6;
    public static final double MAX_RPM = 312;

    public static final boolean RUN_USING_ENCODER = true;
    public static PIDFCoefficients MOTOR_VELO_PID = new PIDFCoefficients(0, 0, 0, getMotorVelocityF(MAX_RPM / 60 * TICKS_PER_REV));

    public static double WHEEL_RADIUS = 1.88976; // in
    public static double GEAR_RATIO = 1.0 / 19.2; // output (wheel) speed / input (motor) speed

    public static double TRACK_WIDTH = 16.5; // in

    public static double kV = 1.0 / rpmToVelocity(MAX_RPM);
    public static double kA = 0;
    public static double kStatic = 0;

    public static double MAX_VEL = 30;
    public static double MAX_ACCEL = 30;
    public static double MAX_ANG_VEL = Math.toRadians(180);
    public static double MAX_ANG_ACCEL = Math.toRadians(180);

NoahBres, 2021-02-22:
Leave gear ratio at 1

MischeviousMasterMind, 2021-02-22:
It is still overshooting.

MischeviousMasterMind, 2021-02-22:
One thing to note is that the distance traveled in the Straight Test remained consistent, so I don't believe it is a problem with the drive constants.

NoahBres, 2021-02-22:
> It is still overshooting.

By how much? If it's a few inches it's probably to be expected due to the decel problem. I don't think anyone has quantified it but I am betting if you integrate the decel phase lag it comes out to the distance that the bot overshoots. Just let the follower pids handle that.

MischeviousMasterMind, 2021-02-22:
It still overshot by about -74 inches, which is way too much for comfort, and the same for the first time.

NoahBres, 2021-02-22:
lol that is quite the error. You also have yet to tune your velocity pid.

AnwarMP, 2021-03-25:
It looks like you have tuned neither the velo PID or the FF, can you update this once you tune on of those

closed 2021-12-15 by rbrott