acmerobotics/road-runner-quickstart #131: Strafing wheels directions aren't correct but turning and straight are correct

SilasBehnke, 2021-02-22:
Basically the bot when we run Straight test it hits it on the head and its nearly perfect, but when we try the strafe test, the left wheels try to strafe left and the right wheels try to strafe right. We had to reverse two motors to drive straight and when we unreversed them strafing worked, but then Straight test just spun in a circle. Is there a way to fix that?

rbrott, 2021-02-22:
Make sure your wheels look like this from the top: [image].

SilasBehnke, 2021-02-22:
they are correct, I have checked. the motor directions themselves just do not spin properly

rbrott, 2021-02-22:
OK then reverse motors until forward is indeed forward and make sure the motor names/assignments are correct.

SilasBehnke, 2021-02-23:
that's what I did, but as a result, the strafing doesn't work. It seems to be either or

rbrott, 2021-02-23:
Not really sure what to tell you then. I would be extra systematic and check each wheel independently.

closed 2021-02-23 by rbrott

Jarvx404, 2022-12-29:
Hey there! Did this issue get solved because I'm having the exact same problem now :/

SilasBehnke, 2022-12-29:
Yea so actually this isn’t a software issue, at least in our case, it’s an issue with hardware, iirc you have a Left and Right motor flipped in your control hub, fixing that *should* fix any problems you have. 

TL;DR: Double Check your cable runs, rerun cables if you need to.

Jarvx404, 2022-12-31:
> Yea so actually this isn’t a software issue, at least in our case, it’s an issue with hardware, iirc you have a Left and Right motor flipped in your control hub, fixing that _should_ fix any problems you have.
> 
> TL;DR: Double Check your cable runs, rerun cables if you need to.

Yeah, that was also the case for us :)