acmerobotics/road-runner-quickstart #135: Dashboard disable

Team14423, 2021-03-09:
Hi -

We have update to the latest version (including .0.4.0 in gradle for Dashboard) but the disable dashboard opmode is not showing up for us. Where do we look for that? Do we need the driver station to be at 6.2 for that? We haven't upgraded that yet.

rbrott, 2021-03-09:
The latest version has dash 0.4.1. I suspect upgrading will fix the issue.

Team14423, 2021-03-09:
No luck. We're pulling from Noah's advanced quick start branch. Maybe it's a revision behind, though our code seems to match the instructions on the dashboard install readme.

NoahBres, 2021-03-09:
That branch used `0.3.10`. It has now been updated to `0.4.1`. Apologies for the inconvenience.

Team14423, 2021-03-09:
OK, so pull an update and all good?

Team14423, 2021-03-10:
Thanks all - that worked, all good

closed 2021-03-10 by Team14423