acmerobotics/road-runner-quickstart #139: Turn PIDF

Team14423, 2021-04-01:
Back for more on TurnAsync. Our 3 wheel localizer is inaccurate for reasons we have been unable to debug, so we are going to test out an analog gyro with two wheel localizer. So far, it seems to be working, but for whatever reason turning continues to give us problems, as we added weight to the front of the robot and the profile just doesn't work and our post-profile code is not great. We decided to write our own PID based code, abandoning the profile altogether. The team figured that at least that way they would be familiar with the code and able to tweak it.

But in doing so and using the PIDFController in roadrunner, it occurred to me that the quickstart code may have a bug in the PID Correction. The code continually changes the target based on the profile, but it does not call:
`turncontroller.reset()`
when that target changes between profile slices. I think that's fine for the middle part of the profile, as the target doesn't change for a while, but at the end, when there's a deceleration, you have a very small error, but a very large cumulative error would add up.

But then I looked more closely at the PIDF code, and the I term only uses the sum of the last two errors, and not the cumulative sum of errors. That's not how I would expect an I term to work, which could be one reason why the code is not behaving like we thought, but I guess it does explain why kStatic is so important. 

Am I missing something here? Is the idea that `kStatic` just eliminates the need for a cumulative error? We aren't using `kStatic` (I don't think), except our new PID code uses `setDriveSignal`, which I think has `kStatic` embedded in the `setMotorPowers` portion. So, the question is whether we should be setting `kStatic` in the PIDFController or not (and whether the non-cumulative I is by design)

rbrott, 2021-04-01:
> But in doing so and using the PIDFController in roadrunner, it occurred to me that the quickstart code may have a bug in the PID Correction. The code continually changes the target based on the profile, but it does not call:
> `turncontroller.reset()`

Yes, that's intentional. The integral sum should accumulate over the entire turn. However, there very well may be a bug where `turnController.reset()` isn't called between turns.

> But then I looked more closely at the PIDF code, and the I term only uses the sum of the last two errors, and not the cumulative sum of errors.

That's not quite correct. Each `update()` call adds 1/2(last error + current error) to the running total in accordance with the trapezoidal rule.

> Is the idea that `kStatic` just eliminates the need for a cumulative error?

Well hopefully the feedforward does most of the work. Using I is relatively uncommon in these position loops.

> We aren't using `kStatic` (I don't think), except our new PID code uses `setDriveSignal`, which I think has `kStatic` embedded in the `setMotorPowers` portion. So, the question is whether we should be setting `kStatic` in the PIDFController or not (and whether the non-cumulative I is by design)

The `setMotorPowers()` method directly sets the powers without any adjustments. The feedforward model is designed to take velocity and acceleration to power, and you can tell when it's applied internally based on whether the method accepts velocities/accelerations or powers.

Team14423, 2021-04-01:
>  The integral sum should accumulate over the entire turn.

Whoops! missed the + before the = for errorsum. 

>However, there very well may be a bug where turnController.reset() isn't called between turns.

I think that's worth a look. We're changing it. But I also wonder about between slices of the profile. At the end of the profile, the errors (and sum) should be small, because it's the last deceleration bit, but the sum for I will include all the baggage from earlier. That may be what you want, but if you are building deceleration into the profile, maybe not.

The code now is pretty simple. We set one heading target, and then:
```
double correction = turnController.update(currentPose.getHeading());
if (Math.abs(turnController.getLastError)>Math.toRadians(1)) {
setDriveSignal(new DriveSignal(new Pose2d(0, 0, correction),
                            new Pose2d(0, 0, 0)));}

```
If my math is right, a P of 8.333 and our Kv of .0125 and track width of 12.25 will yield motor powers=1 for any error above 45 degrees, and then go down to motor power of .12 (plus `kStatic`) at 1 degree. At least I hope so.

rbrott, 2021-04-01:
> But I also wonder about between slices of the profile.

That's a reasonable feedback approach among many.

> If my math is right, a P of 8.333 and our Kv of .0125 and track width of 12.25 will yield motor powers=1 for any error above 45 degrees, and then go down to motor power of .12 (plus `kStatic`) at 1 degree. At least I hope so.

I haven't checked your math, but you should be able to test offline without too much difficulty.

Team14423, 2021-04-01:
>I haven't checked your math, but you should be able to test offline without too much difficulty.

You mean like a simulator? That would be cool. Or just logging the values in my own looping code?

rbrott, 2021-04-01:
I meant run your snippet with a mock `MecanumDrive` that logs motor powers. You should then be able to run it locally without hardware.

Team14423, 2021-04-01:
OK, thanks - that what I thought you meant.

EddieDL, 2021-04-18:
@Team14423 what did you find related to turning? Were there changes you needed to make to get things to work correctly?

Team14423, 2021-04-18:
Turning small distances is still troublesome. We never got the PID to work
well (yet), so we are living with basically our modified code (posted in
another github issue) plus we now reset the controller once when we first
call turnasync. Now that our league tournament is over, we are going to
turn more attention to this, as a precise turn would save us a decent chunk
of time. That said, our improvements have us hitting what we want to hit
90% of the time.

EddieDL, 2021-04-18:
What issue is your code posted into? I would like to see what you changed.

Team14423, 2021-04-18:
123

closed 2021-12-15 by rbrott