acmerobotics/road-runner-quickstart #140: RoadRunner Markers Causing Crashing of Code

timmyjr11, 2021-04-15:
We use Roadrunner for our Autonomous program, and the markers are causing the code to end/crash

For a better understanding what is going on, I'm going to give an high-level overview of the automous, then more information on the problem at hand, and I did to try to fix it:

Overview of automous:

- Moves to the first Powershot on the far left of the goal, using temporal markers to both start the shooting motors and shoot

- Moves right to hit the middle power shot, using a temporal marker to prepare to shoot by reeling back the tapper (pushes the ring to the shooters), then a displacement marker to shoot, does the same thing for the last PowerShot on the right

- Goes to place the wobble goal in the correct square based on the ring configuration and places it, using temporal markers to control the wobble arm that places the wobble goal

- Turns 180° and Moves to the rings of there are 1 or 4 rings present, using displacement markers to prepare to intake

- Collects the rings, using temporal markers to start motors

- Turns back towards the goal and prepares to shoot, using a temporal marker to prepare to shoot

THIS IS THE PART WHERE IT MESSES UP:

- We use a temporal marker to shoot, this Trejectory has 8 total temporal markers, but only two of them are used and then the code ends/crashes when it is not supposed to.

- The robot is then supposed to park afterwards, which it doesn't because it already ended for some reason

Things to know:

1. All the markers were called and used properly until it tries to shoot in the goal, that's when it ends

2. It always ends on the same marker, after the second marker in the shooting Trejectory, no matter how much I change the overall time and the time between, but it still ends on the same part and the same marker.

3. I tried to separate the amount of markers in half, having the second marker move 0.5 inches to the right to satisfy the "EmptyList" Exception, but it still ends on the same point and marker.

4. All the Trajectorys uses ".followTrejectory()" DOESN'T USE ASYNC, meaning it should follow through all the markers regardless if the robot is moving or not, right? 

5. I tried to isolate the code into parts to see what's wrong, every Trjectory works perfectly, markers and all, only the shooting Trejectory is the one that's crashing, meaning there is no outside code that is causing this.

6. I restarted the phone, the robot, and reran the code, but nothing fixed it

7. I even wrote a new .Java file to see if it fixes the problem, it didn't

8. I tried to switch to displacement markers to see if if would fix the problem, it did not, it still ended on the same exact the point the temporal markers did

9. Instead of using markers, I tried to create my own shooting method using a FTC timer Instead of Roadrunners, like this:

tapper.setPosition(1);

timer.reset;

while(timer.miliseconds < 150) {

}

Which does actually shoot the rings without ending the code and crashing it, but since it's not from Roadrunner, it cannot calculate and correct it's angle, meaning it's off angle and misses it's shots.

I don't know how to fix this. I even asked the FTC discord for advice, but it was to no avail. I'm in a major roadblock until fix the problem. so if ANYONE has an idea or solution or a resource to help me with my problem, that would be most appreciated!

rbrott, 2021-04-15:
> * but only two of them are used and then the code ends/crashes when it is not supposed to.

Do you have the stack trace/error message for the crash?

> ```java
>                 .addDisplacementMarker(() -> {
>                     timer.reset();
>                     while (timer.milliseconds() < 250) {
> 
>                     }
>                 })
>                 .build();
> ```

Marker callbacks should never block for extended periods.

> 2\. It always ends on the same marker, after the second marker in the shooting Trajectory, no matter how much I change the overall time and the time between, it still ends on the same part and the same marker.

> 3\. I tried to separate the number of markers in half, having the second trajectory move 0.5 inches to the right to satisfy the "EmptyList" Exception, but it still ends on the same point and marker AT THE SAME EXACT TIME EVERY TIME, NO MATTER WHAT TIME IS LISTED ON THE TEMPORAL MARKERS.

> 4\. All the Trajectorys uses ".followTrejectory()" DOESN'T USE ASYNC, meaning it should follow through all the markers regardless if the robot is moving or not, right?

Markers are designed to execute while the trajectory is running. When the follower has finished the trajectory, it immediately schedules all remaining markers to execute. It does not stall to respect their given times. For example, if I have a 2.5-second trajectory with markers at 1, 2, 3, and 4 seconds, the first two markers will execute as schedule, while the last two will both execute at 2.5 seconds when the follower finishes (this is complicated slightly by admissible error).

timmyjr11, 2021-04-15:
> Do you have the stack trace/error message for the crash?
>
>
> Marker callbacks should never block for extended periods.
>
>
>
>
> Markers are designed to execute while the trajectory is running. When the [...]

> Do you have the stack trace/error message for the crash?
>
>
> Marker callbacks should never block for extended periods.
>
>
>
>
> Markers are designed to execute while the trajectory is running. When the [...]

I get no error message or anything like that. The code kinda stops as if I pressed on the stop button.

If that's the case, then that's what's wrong with my code, it just runs the rest of the markers instantly and at the same time at the end. Is there any way to have it stop moving and follow through with the times then? Because moving while shooting is a problem. And I understand the problem  with blocking the next trajectory with the while statement, the reason I had to do that is because the rings get stuck inside the intake if rings are picked up so closely together in terms of time, which is why I made it stop. I can't use a sleep method outside of the trajectory because it stops the intake motor as well.

rbrott, 2021-04-15:
> Is there any way to have it stop moving and follow through with the times then?

rbrott, 2021-04-15:
> Is there any way to have it stop moving and follow through with the times then?

The easiest solution is find the duration of the trajectory and separate out the late markers into normal op mode code.

timmyjr11, 2021-04-15:
> > Is there any way to have it stop moving and follow through with the times then?
> 
> The easiest solution is find the duration of the trajectory and separate out the late markers into normal op mode code.

Wouldn't that causes my 9th point? Where if it doesn't check it's Trajectory it becomes off angle?

rbrott, 2021-04-16:
My proposed solution works provided you don't need feedback after the robot stops.

If you need feedback after the follower has finished, you can just continue calling `TrajectoryFollower#update()`. This will require some changes to the drive class (perhaps a new mode?).

timmyjr11, 2021-04-16:
> My proposed solution works provided you don't need feedback after the robot stops.
> 
> If you need feedback after the follower has finished, you can just continue calling `TrajectoryFollower#update()`. This will require some changes to the drive class (perhaps a new mode?).

Okay cooollllll, I'll work with that and learn about it further. I had made a solution by having the robot move little distance REALLY SLOWLY in different Trejectory to have all the markers work properly, thanks for the help!

closed 2021-04-16 by timmyjr11