acmerobotics/road-runner-quickstart #142: Field Centric Drive is not responding to commands properly

timmyjr11, 2021-04-17:
When I mix the Field Centric Drive sample code with my tele-op code allowing actions and servos to be moved, when I start the code it there is no movement, until a second later it goes full speed based on where your left stick input is without stopping, even when there is no input on the controller, what I the problem with my code?

My code:
'''java
package RoadRunnerStuff;

import com.acmerobotics.dashboard.FtcDashboard;
import com.acmerobotics.dashboard.telemetry.MultipleTelemetry;
import com.acmerobotics.roadrunner.geometry.Pose2d;
import com.acmerobotics.roadrunner.geometry.Vector2d;
import com.acmerobotics.roadrunner.trajectory.Trajectory;
import com.acmerobotics.roadrunner.trajectory.constraints.AngularVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.MecanumVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.MinVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.ProfileAccelerationConstraint;
import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;
import com.qualcomm.robotcore.eventloop.opmode.TeleOp;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.DcMotorEx;
import com.qualcomm.robotcore.hardware.DcMotorSimple;
import com.qualcomm.robotcore.hardware.PIDFCoefficients;
import com.qualcomm.robotcore.hardware.Servo;
import com.qualcomm.robotcore.util.ElapsedTime;

import java.util.Arrays;

@TeleOp
public class Hbotv2 extends LinearOpMode {
    DcMotorEx frontLeft;
    DcMotorEx frontRight;
    DcMotorEx backLeft;
    DcMotorEx backRight;
    DcMotorEx backShooter;
    DcMotorEx frontShooter;
    DcMotor convey;
    DcMotor intake;
    Servo stopper;
    Servo tapper;
    Servo leftPivot;
    Servo rightPivot;
    Servo Gripper;
    Servo armPivot;

    ElapsedTime timer = new ElapsedTime();

    PIDFCoefficients PID = DriveConstants.MOTOR_VELO_PID;

    double frontVelocity;
    double backVelocity;
    int shooterPower;

    private final FtcDashboard dashboard = FtcDashboard.getInstance();

    @Override
    public void runOpMode() {

        telemetry = new MultipleTelemetry(telemetry, dashboard.getTelemetry());

        frontLeft = hardwareMap.get(DcMotorEx.class, "frontLeft");
        frontRight = hardwareMap.get(DcMotorEx.class, "frontRight");
        backLeft = hardwareMap.get(DcMotorEx.class, "backLeft");
        backRight = hardwareMap.get(DcMotorEx.class, "backRight");
        backShooter = hardwareMap.get(DcMotorEx.class, "backShooter");
        frontShooter = hardwareMap.get(DcMotorEx.class, "frontShooter");
        stopper = hardwareMap.get(Servo.class, "stopper");
        tapper = hardwareMap.get(Servo.class, "tapper");
        convey = hardwareMap.get(DcMotor.class, "convey");
        intake = hardwareMap.get(DcMotor.class, "intake");
        leftPivot = hardwareMap.get(Servo.class, "leftPivot");
        rightPivot = hardwareMap.get(Servo.class, "rightPivot");
[... 427 lines cut ...]
                    drive.backShooter.setVelocity(1720);

                })

                .addDisplacementMarker(() -> {
                    drive.tapper.setPosition(1);
                    drive.frontShooter.setVelocity(1150);
                    drive.backShooter.setVelocity(1720);
                })

                .build();

        drive.followTrajectory(moveToShootPower);
        drive.followTrajectory(shootPowerTwo);
        drive.followTrajectory(shootPowerThree);

        drive.frontShooter.setVelocity(0);
        drive.backShooter.setVelocity(0);
        drive.armPivot.setPosition(0);
        drive.Gripper.setPosition(0.45);
        shooterPower = 0;

    }
}
'''

EddieDL, 2021-04-18:
Currently you are creating a new `SampleMecanumDrive()` each time you call `driving()`. Not sure if that is your issue, but it can't help.

```
private void driving(){
        SampleMecanumDrive drive = new SampleMecanumDrive(hardwareMap);

        // Read pose
        Pose2d poseEstimate = drive.getPoseEstimate();

        // Create a vector from the gamepad x/y inputs
        // Then, rotate that vector by the inverse of that heading
        Vector2d input = new Vector2d(
                -gamepad1.left_stick_y,
                -gamepad1.left_stick_x
        ).rotated(-poseEstimate.getHeading());

        // Pass in the rotated input + right stick value for rotation
        // Rotation is not part of the rotated input thus must be passed in separately
        drive.setWeightedDrivePower(
                new Pose2d(
                        input.getX(),
                        input.getY(),
                        -gamepad1.right_stick_x
                )
        );

        // Update everything. Odometry. Etc.
        drive.update();
}
```

timmyjr11, 2021-04-18:
> Currently you are creating a new `SampleMecanumDrive()` each time you call `driving()`. Not sure if that is your issue, but it can't help.
>
> ```
> private void driving(){
>         SampleMecanumDrive [...]

I wondered if that was the problem as well, so I created a new code that had the driving method code put inside the while(opModeIsActive()) part, still does the same thing

EddieDL, 2021-04-18:
Putting the driving code into the main `while(opModeIsActive())` won't address the issue I mention. The only way to fix it is to remove the line `SampleMechanumDrive drive = new SampleMechanumDrive(hardwardMap);`.  By removing this you will allow your "driving" code to share the same SampleMechanumDrive object with the rest of the code in your class.

EddieDL, 2021-04-18:
Actually I looked at a few more of your methods, and they all have this same issue. You are creating brand new `SampleMechanumDrive(hardwareMap)` objects in many of your methods. You should only do this one time at the beginning of your code.

timmyjr11, 2021-04-18:
> Actually I looked at a few more of your methods, and they all have this same issue. You are creating brand new `SampleMechanumDrive(hardwareMap)` objects in many of your methods. You should only do this one time at the beginning of your code.

The reasoning for that is because when I create a new method is because if I want to the hardwaremap from `SampleMecanumDrive(hardwareMap)`, I makes me call a new SampleMecanumDrive, because for some reason it won't be able to resolve what `drive.` means when I call a servo for example. Is there anyway to get around that?

Oh and also, for more clarification, I called the new HardwareMap for the reasons above. When I placed the driving method into `while(OpModeIsActive && !isStopRequested)` I removed the `SampleMecanumDrive(hardwareMap)` to make sure the hardwareMap was only called once in `class runOpMode`

EddieDL, 2021-04-19:
Anything you want to be shared by all of your methods you need to move it's declaration from the method to the class. For example you have the following objects defined at the class level. But things like `drive` are not.

[image]

So move your objects you want to share up to that location and then you should be able to share it.

timmyjr11, 2021-04-21:
> Anything you want to be shared by all of your methods you need to move it's declaration from the method to the class. For example you have the following objects defined at the class level. But [...]

So once I did move the hardwareMap into the class rather than a method, I came upon a new problem.

"User code threw an uncaught exception: NullPointerException - Attempt to invoke virtual method 'java.util.List.com.qualcomm.robotcare.hardware.HardwareMap.GetAll(java.lang.Class)' on a null object refference"

This only happened after I moved the hardwareMap into the class rather than the method. I also added to a `throws InterruptedException` to see if that would fix the problem, which it did not.

Code:

```java
package RoadRunnerStuff;

import com.acmerobotics.dashboard.FtcDashboard;
import com.acmerobotics.dashboard.telemetry.MultipleTelemetry;
import com.acmerobotics.roadrunner.geometry.Pose2d;
import com.acmerobotics.roadrunner.geometry.Vector2d;
import com.acmerobotics.roadrunner.trajectory.Trajectory;
import com.acmerobotics.roadrunner.trajectory.constraints.AngularVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.MecanumVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.MinVelocityConstraint;
import com.acmerobotics.roadrunner.trajectory.constraints.ProfileAccelerationConstraint;
import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;
import com.qualcomm.robotcore.eventloop.opmode.TeleOp;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.DcMotorSimple;
import com.qualcomm.robotcore.util.ElapsedTime;

import java.util.Arrays;

@TeleOp
public class Hbotv3 extends LinearOpMode {

    ElapsedTime timer = new ElapsedTime();

    double frontVelocity;
    double backVelocity;
    int shooterPower;

    SampleMecanumDrive drive = new SampleMecanumDrive(hardwareMap);

    private final FtcDashboard dashboard = FtcDashboard.getInstance();

    @Override
    public void runOpMode() throws InterruptedException {

        telemetry = new MultipleTelemetry(telemetry, dashboard.getTelemetry());

        drive.frontShooter.setDirection(DcMotorSimple.Direction.REVERSE);
        drive.backShooter.setDirection(DcMotorSimple.Direction.REVERSE);
        drive.intake.setDirection(DcMotorSimple.Direction.REVERSE);
        drive.frontShooter.setMode(DcMotor.RunMode.RUN_USING_ENCODER);
        drive.backShooter.setMode(DcMotor.RunMode.RUN_USING_ENCODER);

        // Initialize SampleMecanumDrive
        // We want to turn off velocity control for tele-op
        // Velocity control per wheel is not necessary outside of motion profiled ;

        drive.setMode(DcMotor.RunMode.RUN_WITHOUT_ENCODER);

        // Retrieve our pose from the PoseStorage.currentPose static field
        // See AutoTransferPose.java for further details
        drive.setPoseEstimate(PoseStorage.currentPose);

        waitForStart();

        drive.tapper.setPosition(0);
        drive.stopper.setPosition(1);
        drive.leftPivot.setPosition(0);
        drive.rightPivot.setPosition(1);
[... 354 lines cut ...]
                    drive.frontShooter.setVelocity(1150);
                    drive.backShooter.setVelocity(1720);

                })

                .addDisplacementMarker(() -> {
                    drive.tapper.setPosition(1);
                    drive.frontShooter.setVelocity(1150);
                    drive.backShooter.setVelocity(1720);
                })

                .build();

        drive.followTrajectory(moveToShootPower);
        drive.followTrajectory(shootPowerTwo);
        drive.followTrajectory(shootPowerThree);

        drive.frontShooter.setVelocity(0);
        drive.backShooter.setVelocity(0);
        drive.armPivot.setPosition(0);
        drive.Gripper.setPosition(0.45);
        shooterPower = 0;
    }
}
```

closed 2021-12-15 by rbrott