acmerobotics/road-runner-quickstart #143: Problem with the Dashboard.

AbilityEdge, 2021-05-11:
[robotControllerLog.txt](https://github.com/acmerobotics/road-runner-quickstart/files/6461500/robotControllerLog.txt)
Hey!
I was trying to use the Dashboard here, and it wasn't quite working... you can choose and run opmodes on the Driver Station App freely, but when I try to open the Dashboard and want to run an opmode, the dashboard connects and disconnect from the robot very quickly in a loop, and then the robot tries to restart itself but it fails, and restarts and restarts...
here is the logcat from the Control Hub:

rbrott, 2021-05-11:
The key message is `java.lang.IllegalArgumentException: Infinity is not a valid double value as per JSON specification. To override this behavior, use GsonBuilder.serializeSpecialFloatingPointValues() method`.

One of your config variables is infinite; this is usually caused by a zero gear ratio (see the warning in [this LRR section](https://www.learnroadrunner.com/drive-constants.html#wheel-radius-gear-ratio-trackwidth)).

AbilityEdge, 2021-05-12:
> The key message is `java.lang.IllegalArgumentException: Infinity is not a valid double value as per JSON specification. To override this behavior, use [...]

I checked the DriveConstants, and the StandardTrackingWheelLocalizer.... I see no value that is 0 except some of the PIDF values... Where can the problem be?

rbrott, 2021-05-12:
Can you post the code of all files annotated with `@Config`?

NoahBres, 2021-05-12:
@AbilityEdge It's not that your value is zero. It's that it evaluates to a zero because you're representing a fraction using integer division instead of doubles. It's not an inherently intuitive or obvious issue. See the example that Ryan linked for an example of the error:

[image]

EliorYousefi, 2021-05-12:
ohhh, sorry! thank you it works!!!!

closed 2021-12-15 by rbrott