acmerobotics/road-runner-quickstart #144: Problem with StraightTest and StrafeTest

StamatieMihnea, 2021-05-12:
Hey!
When I run one of the mentioned opModes the robot goes in the correct direction but when the error gets closer to 0 (the robot is close to the desired position) it goes in the opposite direction 10-20 inches. I'm using 3 dead wheels for localization and feedForward for velocity control. No matter how much I tune the feedForward parameters, I get the same result. Multiplying the gear ratio or using a lateral_multiplier doesn't seem to help.
Thank you!

rbrott, 2021-05-12:
I'm going to assume you have no follower feedback at this point.

A direction change at the end of a trajectory usually indicates that `kA` is too high.

StamatieMihnea, 2021-05-12:
When i try using a smaller kA it become imposible to tune the velocity graph

closed 2021-12-15 by rbrott