acmerobotics/road-runner-quickstart #154: LocalizationTest OpMode Issue

team16651, 2021-10-08:
The LocalizationTest OpMode doesn't correctly update the values of the telemetry, nor does the dashboard render the movement of the robot correctly.

rbrott, 2021-10-09:
You're going to have to be more specific. The dashboard renders the motion perceived by the odometry system; perhaps there are calibration errors. Rooting out such errors is the main purpose of `LocalizationTest`.

rbrott, 2021-10-09:
There were no substantial changes to the quickstart committed in the v7.0 upgrade. You can view the diff [here](https://github.com/acmerobotics/road-runner-quickstart/compare/82137b6d7c96a6775904840fda45d41c7f6665b0...master). The only major issue I'm aware of [this dashboard one](https://github.com/acmerobotics/ftc-dashboard/issues/67).

team16651, 2021-10-11:
Okay, I feel quite silly here, as it turns out I forgot to set the encoder ticks per rev value, it was zero, thus nothing was being measured. Thank you so much for the hard work you do with this library.

closed 2021-10-11 by team16651