acmerobotics/road-runner-quickstart #156: TrajectorySequenceBuilder addTrajectory followed by turn or wait

shelbyRobots, 2021-11-09:
When you are building a trajectory sequence using add*Marker, lineTo, splineTo, turn, wait, etc.  It seems to work fine.
However, if you use addTrajectory, followed by a turn or wait, it jumps to the sequence start point to perform the wait/turn and then back to the spot it was at in the sequence.
I am trying to build a single TrajectorySequence for our auton, but also use some conditionals that seem to work best using traditional Trajectories vs the streamed calls of TSB.
It seems that turn and wait grab their startPose from the prior sequence segment when using any call other than addTrajectory.

NoahBres, 2021-11-09:
Do you mind posting the code for this scenario?

NoahBres, 2021-11-09:
I believe the problem is that the `addTrajectory()` function does not properly update the state of the builder:

https://github.com/acmerobotics/road-runner-quickstart/blob/82137b6d7c96a6775904840fda45d41c7f6665b0/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/trajectorysequence/TrajectorySequenceBuilder.java#L467

It should follow the same logic as the `addPath()` function and modify the `lastPose`, `currentDuration`, and `currentDisplacement`

https://github.com/acmerobotics/road-runner-quickstart/blob/82137b6d7c96a6775904840fda45d41c7f6665b0/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/trajectorysequence/TrajectorySequenceBuilder.java#L298

I believe modifying the `addTrajectory()` function on line 467 in `TrajectorySequenceBuilder.java` to something like so should ameliorate the issue?

```java
public TrajectorySequenceBuilder addTrajectory(Trajectory trajectory) {
    pushPath();

    sequenceSegments.add(new TrajectorySegment(trajectory));

    lastPose = trajectory.end();
    currentDuration += trajectory.duration();
    currentDisplacement += trajectory.getPath().length();
    
    return this;
}
```

This is off the top of my head and I will have to test in an actual environment to confirm.

Apologies for the oversight. TrajectorySequence wasn't really designed with conditional trajectories in mind and the assumption was that would be be called once.

shelbyRobots, 2021-11-09:
No apology needed.  I appreciate the excellent work you did on this.
I had looked at TrajectorySequenceBuilder and already come to the same conclusion.
I was going to copy the stuff at the end of addPath to the addTrajectory - like you have above.
I didn't trace enough to know if lastDurationTraj and lastDiplacementTraj also need to be updated with the added trajectories time and length.
But - while I have a copy of trajectorysequence directory under our Teamcode, I was doing testing in MeepMeepTesting - which is in a parallel directory, and accesses the sequence stuff as a library I am guessing - because I can't edit the kt file.
When I get a chance I will try the change in our Teamcode module.  Or see if I can get MeepMeep to build in its own project/clone.

NoahBres, 2021-11-14:
> I appreciate the excellent work you did on this.

Thank you! I really appreciate it 😄 

>  didn't trace enough to know if lastDurationTraj and lastDiplacementTraj also need to be updated with the added trajectories time and length

I would have to sit down and review the logic to confirm this. I haven't looked at that code in quite a while. I'll get back to you on this when I have the time if you haven't already tested/figured this out.

> Or see if I can get MeepMeep to build in its own project/clone

It is relatively trivial to get a local build of `MeepMeep` running if you intend to modify it. I'll make note to add directions for this soon. But if you're familiar with the CLI, it's simply a matter of cloning the [MeepMeep repo](https://github.com/NoahBres/MeepMeep) and running `./gradlew build; ./gradlew publishToMavenLocal`. Then, the dependency in your `MeepMeepTesting` module should be changed to `com.github.NoahBres:MeepMeep:1.0-SNAPSHOT`. Apologies for the informal direction here. 

> that I came across this when trying to come up with a way to have a repeated portion of a sequence without copy pasting

Personally, I am of the opinion that explicit copy-pasting is ideal as it represents the behavior of your auto program. If there is a repeated looping portion that you'd like handled, it would be better served by controlling the flow of the trajectories with your own logic (see the state machine pattern). `TrajectorySequency` is meant to be static at "build" time (although it is local code; you are free to bend it to your will). 
I am somewhat against the overzealous application of the DRY principle, especially when the snippets of code in question may happen to be the same but are not explicitly linked/reusable in behavior.
However, I may simply be naïve in this view and Ryan may shed some more experienced opinions on this. But as it is now, I see nothing wrong with copy-pasting repeated segments as they still represent two different explicit movements in time.

shelbyRobots, 2021-11-15:
The main issue with copy paste is maintainability - you have to make changes in multiple places. We are currently running a basic state machine wrapped around a list of TrajectorySequences.  Sometimes more than one of the states are mapped to the same sequence - so we can just cycle.  And we can build all of the sequences, and then choose to remove some of them from our list if the total time is too long, etc.  We also have several routes - for different starts, etc.  And we now use some of the same sequences in multiple routes.
This works out nicely in our Teamcode, but we can only do one seq in MeepMeep.  I copy/pasted code from Teamcode to our MeepTesting.java file - and addressed missing variables and such.  Then I create a new TSB and run through a loop on all of the sequences in our list - adding each segment to the new TSB, and then building an overall sequence for MeepMeep visualization.  That is where I ran into the turn/wait issue.  It is still very helpful and useful - but the robot jumps back to the start position in MeepMeep at each wait/turn.
Btw - doing the rebuild of a new TSB helped us find some marker offset issues that we had missed.  So again - useful.
It would be **_really_** nice to be able to use TSBs or sequences from our Teamcode classes directly in MeepMeep, but I kept running into conflicts since MeepMeep is using separate kotlin definitions of sequence, etc., while Teamcode is using a java trajectorySequence folder copied from LRR.  Is there a clean way to do this?
I would like to construct an instance of our FfRoute class in MeepTesting, and then get a sequence (or list of sequences) from it.  Lists would be reconstructed into a single sequence as described above.  Then I wouldn't have to try to keep updates made when debugging routes in MM back to Teamcode.

shelbyRobots, 2021-11-15:
I'm not strong on gradle.  What does ./gradlew publishToMavenLocal do?

shelbyRobots, 2021-11-15:
Ohh - and BTW - we were using MeepMeep at our qualifier Saturday to show other teams what are routes were.
Several teams were salivating over it, and I know a few are going to be looking into it.

NoahBres, 2021-11-15:
I neglected one more step. In your `repositories` block, you also include `mavenLocal()` below the two urls you have listed.

What `./gradlew publishToMavenLocal` does is build the library and stores the artifacts in your local maven folder alongside the rest of the cached libraries. You can find this on `~/.m2` on macOS. I believe the paths are similar for Linux. Unsure on Windows. And then declaring `mavenLocal()` in your `repositories` block sets this as a source. This also makes development much easier as the local maven repo isn't cached so you can continuously build.

shelbyRobots, 2021-11-16:
@NoahBres Thanks again.  I have tested the fix, and it seems to work.  I will make a pull request.
But I have a few questions.  We are not using addTrajectory in our actual opMode.  There, we use a state machine managed list of sequences built "normally".  But we wanted to be able to use MeepMeep to visualize the **whole** route - so in MeepMeepTest, we loop on the sequences in the list and build a composite sequence.  This is where we found the issue.  But I think that the fix should be made to the TrajectorySequenceBuilder in both MeepMeep and Roadrunner-quickstart.

1. Is there a reason there is both a kotlin and java version of trajectorySequence?
2. Are there any functional differences between the two?
3. The interface is different between the two - making it clunky to copy code from RRQS based teamcode to MM.  For example, the ctor of TSB takes different parameters.  Does it need to be this way for some reason?
4. I _think_ that if MM used the same trajectorySequence pkg as RRQS, it might be able to be integrated onto user's teamcode and then given a trajectorySequence (or list thereof) for easy debugging of routes.  Is this possible?  Can one AS project have both java and kotlin source code - allowing there to be one version of trajectorySequence between RRQS and MM?  If not, can trajectorySequence be separated into its own library or such?
Hopefully this doesn't come across as complaining.  The intent is quite the opposite.  We really like the benefits of trajectorySequences (markers and automatic handling of discontinuities) and MeepMeep has been a huge improvement for our route development and debugging - and allowed us to collaborate with alliance partners at competition.  But the copy/paste/modify of potentially complex sequences between MM and teamcode can lead to errors in the heat of battle.

NoahBres, 2021-11-17:
> Btw - doing the rebuild of a new TSB helped us find some marker offset issues that we had missed. So again - useful.
It would be really nice to be able to use TSBs or sequences from our Teamcode classes directly in MeepMeep, but I kept running into conflicts since MeepMeep is using separate kotlin definitions of sequence, etc., while Teamcode is using a java trajectorySequence folder copied from LRR. Is there a clean way to do this?
I would like to construct an instance of our FfRoute class in MeepTesting, and then get a sequence (or list of sequences) from it. Lists would be reconstructed into a single sequence as described above. Then I wouldn't have to try to keep updates made when debugging routes in MM back to Teamcode.

I originally hoped for this but could not find a clean way to make this happen. TrajectorySequence requires quickstart-specific code: specifically the turns. It cannot be integrated into the Road Runner library itself. The quickstart copy of TrajectorySequence has to be present. In addition to that, the TeamCode module cannot properly be imported into another module. This makes it impossible for MeepMeep and the quickstart to share a package for TrajectorySequence. The difference of Kotlin and Java doesn't matter. It is because they are different packages. 
As far as I am aware, there is not a clean solution to this problem. There is a possibility of creating a library solely for TrajectorySequence that supports arbitrary, dynamic events and with enough wrappers and configuration it could share code between the quickstart and MeepMeep. However, this would not be part of RoadRunner as it departs from the motion profiling core and I would have to maintain my own version of the quickstart—not a sustainable plan.

> Ohh - and BTW - we were using MeepMeep at our qualifier Saturday to show other teams what are routes were.
Several teams were salivating over it, and I know a few are going to be looking into it.

🔥 🔥 

> But I think that the fix should be made to the TrajectorySequenceBuilder in both MeepMeep and Roadrunner-quickstart

Absolutely.

> Is there a reason there is both a kotlin and java version of trajectorySequence?

The Kotlin version exists because MeepMeep is written in Kotlin and because I enjoy writing in it. It could just be written in Java of course but I find it relatively trivial to develop in Kotlin and port over the Java. The quickstart port is in Java as Kotlin requires some minor extra configuration and the quickstart is meant to be kept as close as possible to the FTC provided repo. 

> Are there any functional differences between the two?

No it is a direct port with little difference. 

> The interface is different between the two - making it clunky to copy code from RRQS based teamcode to MM. For example, the ctor of TSB takes different parameters. Does it need to be this way for some reason?

This is because the primary interface for TrajectorySequence in the quickstart is through the `SampleMecanumDrive` helper function.
See: https://github.com/acmerobotics/road-runner-quickstart/blob/1a52698a48594d65ffca71066928b0466eddbdd0/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/SampleMecanumDrive.java#L144

MeepMeep is designed so you can copy-paste this. 
This is demonstrated at around 11:00 of this video: https://youtu.be/vdn1v404go8?t=660

Most users will seldom call the constructor directly.

 > I think that if MM used the same trajectorySequence pkg as RRQS, it might be able to be integrated onto user's teamcode and then given a trajectorySequence (or list thereof) for easy debugging of routes. Is this possible? Can one AS project have both java and kotlin source code - allowing there to be one version of trajectorySequence between RRQS and MM? If not, can trajectorySequence be separated into its own library or such?
 
 Not without a convoluted setup as explained earlier.
 
 > Hopefully this doesn't come across as complaining
 
 Of course not! I am always open to feedback and new views. I know that I am still young and inexperienced in my software career and there is still much to learn. TrajectorySequence was written by a high-schooler and thus is high-school level code. There is always room for improvement.
 
 > But the copy/paste/modify of potentially complex sequences between MM and teamcode can lead to errors in the heat of battle
 
The current version of MeepMeep is meant to be copy-pasteable (as demoed in the video) so you are deviating from the original use case :P
Unfortunately, MeepMeep does only visualize one TrajectorySequence at a time so visualizing multiple, dynamic/complex trajectories is not possible as you've discovered. I did intend to work on this feature. However, I started making a web version of MeepMeep as it would allow me to build much more complex and useful interfaces at a much faster speed (as my experience is primarily in web development). Work on that, and MeepMeep has stalled as I have graduated and am in university. There is much less incentive to put hours into FTC as I have moved on and can be making money writing software instead. My apologies.

shelbyRobots, 2021-11-24:
@NoahBres Loosely associated question...
In TrajectorySequenceFollower, I added a log of the current poseEstimate and lastPoseError (via follower.getLastError()). 
PoseEstimate looks decent, but it looks like the X and Y are flipped on the lastPoseError.  Is there a different orientation convention between poseEstimate and poseError somewhere between RR and LRR trajectorySequenceFollower?

shelbyRobots, 2021-11-24:
@NoahBres 
Loosely related question..
I am using trajectorySequences in RoadRunner.
I added a log print to the isFollowing portion of trajectorySequenceFollower.update.
lastPoseError = follower.getLastError();
                    RobotLog.dd(TAG, "  at %s error %s",
                        poseEstimate, lastPoseError);
 it seems like the X and Y are flipped on the poseError vs the poseEstimate.
At/near the end of a move, the bot should be at X=-56.0, Y=-61.0, H=90.
Here is what I get:
at (-55.297, -61.120, 90.700°) error (0.129, 0.701, -0.700°).
Note that if you offset the poseEst.X by error.Y, you get basically -56.  And if you offset poseEst.Y by error.X, you get basically -61.0.
Any thoughts on why?

NoahBres, 2022-01-31:
Apologies for the really really late reply. Life has been at a stand-still for a few months. Is this still an issue? The unexpected pose?

NoahBres, 2022-01-31:
Also, may I solicit a sample trajectory that you are using? I feel like this may lead to some unexpected behavior regarding the tangent state in the builder. I would like to try and clear that up

rbrott, 2022-08-19:
Mitigated in `quickstart2`

closed 2022-08-19 by rbrott