acmerobotics/road-runner-quickstart #158: QuickStart changes for new season? And documentation for it?

Team14423, 2021-11-14:
Hi -

We're about to start our drivetrain programming and consider RoadRunner indispensable, especially this year, when we have to rely on motor encoders rather than deadwheels. 

We had been using Noah's advanced fork of the quickstart, but have also modified things a bit, especially in the turns in SampleMeccanumDrive (you may recall prior issues on that). 

We like the rest of our robot code setup to start with, and would like to use that as a starting point, but Noah's code isn't updated. So, what should we do? Update from this project?

Also, we're wondering if there is documentation on the new changes to Trajectory - I see the last issue on TrajectorySequence problem, and it looks like that's how the async is now done. I looked at the online site but didn't see anything, so any feedback appreciated.

Thanks!

NoahBres, 2021-11-14:
My fork of the quickstart only serves to host the [`advance-examples` branch](https://github.com/NoahBres/road-runner-quickstart/tree/advanced-examples). This branch hosts some fun experiments that were especially useful in the Ultimate Goal season. The `master` branch of my fork serves no purpose and is just kept up-to-date mirroring this quickstart—ideally (my apologies). This branch from `acmerobotics` should serve as the starting point for all teams.
Apologies for the lack of clarification.

> I see the last issue on TrajectorySequence problem

That is a relatively niche problem in my opinion as conditionally adding multiple trajectories was not the intended use case of `TrajectorySequence`. This use case should be handled with ones own state handling imo. This specific problem should not concern the majority of teams.

>I looked at the online site but didn't see anything, so any feedback appreciated.

LRR does have a section on `TrajectorySequence` (https://learnroadrunner.com/trajectory-sequence.html) but I do recognize that is was thrown together quite haphazardly and I do have intentions to rewrite it. My apologies.

Team14423, 2021-11-15:
Thanks for the feedback. We are forked from advanced examples. Our plan was to just branch from there for the new season. Is your advanced examples up to date with SDK 7.0, etc.? It didn't seem to be last I looked, but maybe I just missed it..

Team14423, 2021-11-15:
Well, dang, I don't know how I missed an entire section on the LRR page.

NoahBres, 2021-11-15:
> Thanks for the feedback. We are forked from advanced examples. Our plan was to just branch from there for the new season. Is your advanced examples up to date with SDK 7.0, etc.? It didn't seem to be last I looked, but maybe I just missed it..

It is not. My apologies. I have been behind on that. It was never really intended to be forked from. Just a repo to copy-paste code snippets. Updating it has been in my todo-list though.

closed 2021-12-15 by rbrott