acmerobotics/road-runner-quickstart #163: RFC - TrajectorySequence switch from continuous building

NoahBres, 2022-03-02:
Currently, `TrajectorySequence` will continuously build and split trajectories on each builder command.

This was a naive design choice on my part. I would like to propose simply appending the sequence segments in a list and performing the entire build process in the `TrajectorySequenceBuilder#build()` method. Currently, the `build()` command only serves to project the markers back onto the split trajectories.

The impetus for this design decision is for a number of people requesting a `TrajectorySequenceBuilder#mirror()` method, as most FTC games will be symmetrical on an axis. This allows for reusable trajectories for red/blue sides. Currently, the solution for this would be quite unpleasant. It would a much better design if builder had access to the original commands during the build process.

You may decide that a mirror function is unnecessary and better solved individually by the users. In which case, I will simply provide a gist solution for it elsewhere. Nonetheless, I still think this proposal would be a better design decision.

rbrott, 2022-03-05:
I think some kind of pose map function is reasonable. But should it be passed to the constructor or a `map()`/`mirror()` function? The former option further overloads the constructor. It's at the point where another builder is warranted. However, this leaves the semantics clear. One function is applied to all poses in the trajectory. The latter option spares the constructor at the cost of more complicated semantics. Either the last function provided is applied to all poses or each `map()` call applies to all following poses in builder calls until the next `map()`. Only one of these choices requires deferring all trajectory generation to build time.

The same functionality would also be useful for `TrajectoryBuilder`/`PathBuilder`. The composition should be relatively clear.

NoahBres, 2022-03-05:
The overarching composition story you described was very elegant and not something I thought of. I love the pose map solution.

I arrived at the same design predicaments you stated. I am not a fan of a transformation method as a builder method. The semantics are not inherently obvious, as you stated. I am of the opinion that there should be a single transformation applied at build time. Having local contexts is already a behavior exhibited in `TrajectorySequenceBuilder` (constraints and reverse) so there is precedence for this. However, I believe that pose transformations would introduce unnecessary complexity that would be seldom used.

> Either the last function provided is applied to all poses

This behavior is most clear in my opinion. But I still do not fancy the idea of allowing the user to insert multiple, inert `map()` calls throughout the builder.

Passing a mapping method into the constructor would be the best of the options provided. But this forces you to copy-paste entire builders to utilize transformed sequences.

I think a better solution would be to pass in an overload to the build function itself. This allows one to declare a builder once and return two separate sequences, avoiding code duplication.

```java
TrajectorySequenceBuilder builder1 = TrajectorySequenceBuilder()
    .forward(10)
    .splineTo();

TrajectorySequence sequenceRed = builder1.build()

TrajectorySequence sequenceBlue = builder1.build(pose -> {
    // Perform transformation (e.g. mirror)
    return pose;
})
```

rbrott, 2022-03-05:
I like the idea of passing the arguments to `build()`. I took a minute to code up a [proof-of-concept implementation for `PathBuilder`](https://github.com/acmerobotics/road-runner/blob/8239fa86301574a90a0c91bb8c0c42ca811d3c41/core/src/main/kotlin/com/acmerobotics/roadrunner/path/PathBuilder.kt).

rbrott, 2022-08-19:
While I aspire to [aphyr's ideal](https://aphyr.com/posts/321-builders-vs-option-maps), I'll implement this in 0.6.x with polymorphism.

closed 2022-08-19 by rbrott