acmerobotics/road-runner-quickstart #165: `TrajectorySequenceBuilder#setConstraints` does not take into affect until the next trajectory

NoahBres, 2022-03-12:
[`TrajectorySequenceBuilder#setConstraints`](https://github.com/acmerobotics/road-runner-quickstart/blob/b2555fdc164fe1a233b016d7425aab7c9cfa5fce/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/trajectorysequence/TrajectorySequenceBuilder.java#L330) (or any of the helper constraints functions) does not break the current trajectory (call `TrajectorySequenceBuilder#newPath`), thus the expected changes will not be reflected until a new path is automatically created via discontinuity. 

I am not a fan of this implicit behavior; the aforementioned helper functions should force the trajectory sequence builder to break its current trajectory.

rbrott, 2022-03-12:
Not sure about breaking the current trajectory, but the constraints should definitely be applied per path segment.

rbrott, 2022-08-19:
Mitigate in `quickstart2`

closed 2022-08-19 by rbrott