acmerobotics/road-runner-quickstart #166: Acmerobotics URL is offline

developer-Pavel, 2022-03-18:
Cannot get to build road-runner, the plugin website is offline.

> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Could not determine the dependencies of task ':TeamCode:compileDebugJavaWithJavac'.
> > Could not resolve all task dependencies for configuration ':TeamCode:debugCompileClasspath'.
>    > Could not resolve com.acmerobotics.roadrunner:core:0.5.5.
>      Required by:
>          project :TeamCode
>       > Could not resolve com.acmerobotics.roadrunner:core:0.5.5.
>          > Could not get resource 'https://maven.brott.dev/com/acmerobotics/roadrunner/core/0.5.5/core-0.5.5.pom'.
>             > Could not GET 'https://acmerobotics.z13.web.core.windows.net/com/acmerobotics/roadrunner/core/0.5.5/core-0.5.5.pom'.
>                > Connect to acmerobotics.z13.web.core.windows.net:443 [acmerobotics.z13.web.core.windows.net/127.0.0.1] failed: Connection refused: connect
>

rbrott, 2022-03-18:
I've seen some intermittent failures before. The link is up for me now.

developer-Pavel, 2022-03-19:
@rbrott Yes, now it's online, but when I opened the issue I cannot open it even from a web browser, so it wasn't an IDE problem. Anyway, it's fixed now, thank you for your reply.

closed 2022-03-19 by developer-Pavel