acmerobotics/road-runner-quickstart #167: RFC - Consider Removing `GEAR_RATIO` in `DriveConstants.java`

NoahBres, 2022-03-20:
Proposal:

- Consider removing `GEAR_RATIO` in the `DriveConstants.java` file 
- Leave `TICKS_PER_REV` and `MAX_RPM` to be the final calculation without any assumption of a gearbox between the motor and wheel. They should adjust `TICKS_PER_REV` accordingly

Reasoning:
- It is very very common for people to misunderstand what `GEAR_RATIO` is doing. One frequent instance is students putting down something like `19.2` for something like a `19.2` motor as that is how goBILDA labels their Yellowjackets (https://www.gobilda.com/yellow-jacket-planetary-gear-motors/). However, the `TICKS_PER_REV`and `MAX_RPM` are generally already take this ratio off the bare motor into account
- One needs to take into account where the encoder is located along the gear train to actually use different numbers for these. Forcing the user to user to just use the final `TICKS_PER_REV` and `MAX_RPM` should simplify this interface
- It is also a very common point of confusion to swap `GEAR_RATIO` from `OUTPUT / INPUT` to `INPUT / OUTPUT`. I do not know the order off the top of my head either

NoahBres, 2022-03-20:
Alternatively:
- `EXTERNAL_GEAR_RATIO`
- Set `TICKS_PER_REV` to 28 and `MAX_RPM` to 6000 (for the standard FTC bare motor) and have a `TOTAL_GEAR_RATIO` property
    - I believe Hex Core motors are the only one not to follow this?

rbrott, 2022-03-20:
Gear ratio should go to eliminate `double x = 1 / 20` mistakes alone. 

I'm broadly sympathetic to collapsing redundant parameters. RR only need a few gadgets: one that takes velocity+acceleration to motor commands, one that takes encoder readings to distances, and one that converts between angular displacement/velocity and linear displacement/velocity. I think the first two are best captured with a trio of max RPM (rev/min), ticks/rev, and in/rev. Users can choose any shaft in the gear train to mark a revolution. Then the remaining angular-linear velocity/displacement relationship can be captured with track width. 

As you point out, encoder placement 

As you point out, encoder placement is a good reason.

rbrott, 2022-03-20:
oops

rbrott, 2022-08-19:
Removed in `quickstart2`

closed 2022-08-19 by rbrott