acmerobotics/road-runner-quickstart #171: Publish the FTC-specific code as a Gradle artifact

NoahAndrews, 2022-09-19:
Right now, in order for FTC teams using Road Runner to update to a new SDK release (which is mandated for each new season), they have to either wait for this repository to get updated, or they have to merge in the latest release themselves (which requires knowledge of git remotes and merging). They could also use Road Runner directly without the quickstart, but that seems inadvisable given the amount of code here.

Is there a reason that the code that currently lives here couldn't be moved to a Gradle artifact, which would then be referenced in this repository? That would give users more flexibility in how they update, and also open the door to publishing Road Runner artifacts intended for use in OnBotJava (which has explicit support for external libraries now).

rbrott, 2022-09-21:
Frankly, the current arrangement minimizes the work I need to do when I make a change or merge a new SDK version. It also keeps everything synced together in one place and allows me more freedom in making changes across drive classes and the rest of the code. I haven't seen lack of OBJ support and messy Git histories as major pain points (beyond the size of the repo which I hope to remedy in `quickstart2` soon), though I'm open to being convinced otherwise.

thoughtless42, 2022-09-23:
May I suggest based on how the later SDK use gradle it is possible to put all of the roadrunner specific dependencies  into build.dependencies.gradle within the official 8.0 SDK. having used a bash script on both gitpod and colab to git clone and build the apk, when the team member only had access to chromebooks.  This would not solve keeping in sync with the updates TeamCode folders though. 

echo "Fixing files"
echo "repositories {maven { url = 'https://maven.brott.dev/' }}" >> ./build.dependencies.gradle
echo "dependencies {implementation 'org.apache.commons:commons-math3:3.6.1'}" >> ./build.dependencies.gradle
echo "dependencies {implementation 'com.acmerobotics.roadrunner:core:0.5.6'}" >> ./build.dependencies.gradle
echo "dependencies {implementation 'com.acmerobotics.dashboard:dashboard:0.4.5'}" >> ./build.dependencies.gradle

rbrott, 2023-08-29:
Done, see https://github.com/acmerobotics/road-runner-ftc

closed 2023-08-29 by rbrott