acmerobotics/road-runner-quickstart #178: Dashboard View for DriveVelocityPIDTuner

baqwas, 2022-11-01:
Hello!

Our team is stepping through Quickstart. Unfortunately we're stuck at **DriveVelocityPIDTuner**. When we run this OpMode, the Dashboard view does not match the [illustrated guide](https://learnroadrunner.com/drive-velocity-pid-tuning.html#tuning). Two users reported identical issue (they cloned the same fork:)).

On the Dashboard, drilling down under DriveVelocityTuner, the fields are:
- Distance: which works well in setting the back-and-forth distance
- Unexpected fields: kV, kStatic, etc.
- _Not visible_: the PID coefficients (the ones that we should be changing via Dashboard for course correction)

Also, we're unable to get any graphs to display after the run is completed/stopped. The right pane under Telemetry has values for velocity0..3 but zero for error0..3.

I suspect that this is a newbie issue on our part. How can we get the Dashboard to display the PID coefficients that we need to tune for **DriveVelocityPIDTuner**? Thanks.

Kind regards.

rbrott, 2022-11-02:
> When we run this OpMode, the Dashboard view does not match the [illustrated guide](https://learnroadrunner.com/drive-velocity-pid-tuning.html#tuning).

The dashboard need not match the screenshots exactly.

> Not visible: the PID coefficients (the ones that we should be changing via Dashboard for course correction)

The PID coefficients should be under the `DriveConstants` drop-down.

> Also, we're unable to get any graphs to display after the run is completed/stopped.

That's expected unless you pause before stopping (unfortunately, to unpause in the current version, you must close the graph and re-open it).

baqwas, 2022-11-02:
Thanks a bunch for guidance, @rbrott.

closed 2022-11-02 by baqwas