acmerobotics/road-runner-quickstart #183: quickstart2 - no good way to reverse motor encoders independently of motor

NoahBres, 2022-11-11:
From @abidingabi:

[image]
![CleanShot 2022-11-11 at 1
[image]
8 17 12@2x](https://user-images.githubusercontent.com/6739076/201443203-2e037a6b-2793-4f82-bae3-265d0823543a.jpg)

rbrott, 2022-11-12:
> no good way to reverse motor encoders independently of motor

I intend to allow that in `RawEncoder`. The logic could be more clear (negate iff `m.getDirection() != direction`), though I think it should work to apply the encoder direction independent of the motor direction. 

> you can reverse the direction of `RawEncoder`s but not `OverflowEncoder`s

True, but you can reach through the overflow encoder: `overflowEncoder.encoder.direction = ...;`

abidingabi, 2022-11-12:
Oh, I missed that `encoder` was a public field in `OverflowEncoder` when skimming it trying to figure it how to reverse it; that rectifies the problem in a neat way.

closed 2022-12-04 by NoahBres