acmerobotics/road-runner-quickstart #196: Unable to Reach 180 Degrees on Turn Test

8toAutomate, 2023-01-07:
Hey everyone, we have been having some problems with tuning turning with Roadrunner. We run turn test for 180 degrees but at best get 1.5 degrees undershot error- with a track width of 50 and a max angular velocity of less than half of our actual max angular velocity. We have tried track width values from 5 to 50, and various different max angular velocities, but never seem to get the perfect 180, or even above 180, that Learn Roadrunner says you are supposed to get. We measured the angles manually and found that our localizer measured the error accurately. When we try the automatic track width tuner, we never overshoot past 180 degrees, our target value, and the program keeps suggesting a track width that is about 0.4 higher than the last, up to egregious values. Is there anything we are missing that can allow us to turn accurately while also keeping reasonable speed and track width values? More importantly why can we never reach 180 degrees or above in our turns?

rbrott, 2023-01-08:
It concerns me most that you can't get a turn beyond 180 despite changing max ang vel and track width. I would graph the power values that are sent to the motors to see what the shape of the profile is.

8toAutomate, 2023-01-14:
We have been working on tuning and are adjusting the tuning values. We will get back to you soon. Thank you for your help

litl555, 2023-01-14:
Hi there! We were facing the same issue but we fixed it by redoing our manual feedforward test. Our kV was so low that it was not allowing us to go beyond 180 degrees. Try retuning feedforward as detailed in the documentation. Hope this helps! This was an annoying issue to run into. Best of luck to ya'll.

8toAutomate, 2023-01-18:
We returned and returned the whole thing several times, each time the same results.    The turn test never exceeds 176.4.  Slowing down the max speed by half help get it to 178, but by then, the robot is so slow.  In addition, heading and follower pid has never worked.  The robot drifts left in the forward directions and right in the reverse.  we attached a link to a short video that shows the condition.  The robot will drift sideways a complete tile width in not time, but roadrunner still shows the robot straight.  We found a value that keeps the robot straight but once transnational PID is added, it starts drifting again. 
All dimensions an location of odometry wheels were checked and triple checked.  The Odometry pods have no play whatsoever.  The encoders are putting out proper signal and were checked an oscilloscope. The pods are free to rate on the precision machined shafts (with double ball bearings) and the spring tension keeps the wheels in constant contact with the field.  

Back and forth test video:  https://u.pcloud.link/publink/show?code=XZ2ue9VZWKQdUUjAzUf28L71D6HvyYs6CeRk

We have come to the conclusion that this has to do with the drive motors.  One of the drive motors is weaker in the forward direction but not in reverse and the other is weaker in reverse direction but not forward, leading the robot to drift left in the forward direction and right in reverse.  Using teleop with velocity pid ensures the robot drives straight.  We discussed this issue with GoBilda and Ethan conformed that brushed mot will sometime do that.  replacing the motors might help , but the issue will eventually reappear as the motors wear in.  Indeed, we have a brand new chassis kit and it exhibits a very slight drift in a similar manner.  The chassis were checked for squareness and straightness using high precision machining squares, some costing more than the chassis kit.

We have decided to go back and re-tune using velocity PID.  So far, the robot can get to 180 degrees no problem with a track width that is almost exactly matches the physical track width.  The follower is much better and the square test is decent - not perfect, but decent.  Correct us if we are wrong, but feed forward tuning applies the same power to all motors.  So if some of the motors are lagging, then robot will drift towards one side or another.  Then the training relies heavily on heading pid to make corrections.

One of the remaining issues I mentioned before is the heading PID is not perfect.  The robot will still drift to the side slightly.  Doing the follower PID test with the square test, the robot will drift out of the alignment on the second round, but the FTC dashboard shows the robot is perfectly aligned.   Any idea why?

[image]
[image]

rbrott, 2023-01-19:
> The robot drifts left in the forward directions and right in the reverse. we attached a link to a short video that shows the condition. The robot will drift sideways a complete tile width in not time, but roadrunner still shows the robot straight.

> Back and forth test video: https://u.pcloud.link/publink/show?code=XZ2ue9VZWKQdUUjAzUf28L71D6HvyYs6CeRk

> Doing the follower PID test with the square test, the robot will drift out of the alignment on the second round, but the FTC dashboard shows the robot is perfectly aligned.

This points to a problem with localization/odometry.

> In addition, heading and follower pid has never worked. 

Hard for me to know what "never worked" really means, but I suspect this is a symptom of localization problems.

> Correct us if we are wrong, but feed forward tuning applies the same power to all motors. 

[Correct](https://github.com/acmerobotics/road-runner-quickstart/blob/6585a735d3ab1e4032096785438fd0dd30e594f1/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/opmode/ManualFeedforwardTuner.java#L115)

> So if some of the motors are lagging, then robot will drift towards one side or another. Then the training relies heavily on heading pid to make corrections.

Yes, though you may have disturbances from the velocity PID. There's really no good solution here unfortunately, but you may try using feedforward alone without velocity PID. Hopefully that will avoid exacerbating the inevitable lag.

closed 2023-09-21 by rbrott