acmerobotics/road-runner-quickstart #213: Quickstart Install doesn't appear to work on MacOS

jeremyruppel, 2023-02-23:
Apologies in advance if I'm making a basic gradle mistake. Trying to simply `gradle build` fails on MacOS Monterey with the following error:

```
Git/road-runner-quickstart $ gradle build                                                                                                                                                                                 

> Configure project :FtcRobotController
WARNING:Using flatDir should be avoided because it doesn't support any meta-data formats.

> Configure project :TeamCode
WARNING:Using flatDir should be avoided because it doesn't support any meta-data formats.
WARNING:Using flatDir2 should be avoided because it doesn't support any meta-data formats.

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':FtcRobotController:bundleDebugAar'.
> Could not create task ':FtcRobotController:packageDebugAssets'.
   > Cannot use @TaskAction annotation on method IncrementalTask.taskAction$gradle_core() because interface org.gradle.api.tasks.incremental.IncrementalTaskInputs is not a valid parameter to an action method.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 329ms
```

Here's my gradle and java version.

```
Git/road-runner-quickstart $ gradle -v                                                                                                                                                                                     master * ] 9:16 PM

------------------------------------------------------------
Gradle 8.0.1
------------------------------------------------------------

Build time:   2023-02-17 20:09:48 UTC
Revision:     68959bf76cef4d28c678f2e2085ee84e8647b77a

Kotlin:       1.8.10
Groovy:       3.0.13
Ant:          Apache Ant(TM) version 1.10.11 compiled on July 10 2021
JVM:          19.0.2 (Homebrew 19.0.2)
OS:           Mac OS X 12.6.1 aarch64
```

```
Git/road-runner-quickstart $ java -version                                                                                                                                                                                
openjdk version "19.0.2" 2023-01-17
OpenJDK Runtime Environment Homebrew (build 19.0.2)
OpenJDK 64-Bit Server VM Homebrew (build 19.0.2, mixed mode, sharing)
```

rbrott, 2023-02-23:
Hello! Maybe your Gradle version is too recent? Try `./gradlew build` which uses the project's Gradle version:
```
$ ./gradlew -v

------------------------------------------------------------
Gradle 7.4.2
------------------------------------------------------------

Build time:   2022-03-31 15:25:29 UTC
Revision:     540473b8118064efcc264694cbcaa4b677f61041

Kotlin:       1.5.31
Groovy:       3.0.9
Ant:          Apache Ant(TM) version 1.10.11 compiled on July 10 2021
JVM:          14.0.2 (AdoptOpenJDK 14.0.2+12)
OS:           Mac OS X 10.16 x86_64
```

jeremyruppel, 2023-02-23:
That seems to have done the trick! Thanks!

closed 2023-02-23 by jeremyruppel