acmerobotics/road-runner-quickstart #214: Parameters to pass through setLocalizer(new StandardTrackingWheelLocalizer(hardwareMap, lastTrackingEncPositions, lastTrackingEncVels));

awesomehtr1, 2023-02-23:
Hi,

What parameters do you pass in for lastTrackingEncPositions and lastTrackingEncVels? I haven't seen this before and don't know what to pass in for this. Thanks in advance!

rbrott, 2023-02-23:
You can look at [what `SampleMecanumDrive` does](https://github.com/acmerobotics/road-runner-quickstart/blob/563b195593b5bdf0f3dc38ec509fdade29844438/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/SampleMecanumDrive.java#L131).

awesomehtr1, 2023-02-23:
How would you figure out "lastTrackingEncPositions" and "lastTrackingEncVels"? That's the part I don't understand. Thank you.

awesomehtr1, 2023-02-23:
Nevermind, I was being dumb. I can just set it as "setLocalizer(new StandardTrackingWheelLocalizer(hardwareMap,lastTrackingEncPositions,lastTrackingEncVels));" Thank you.

closed 2023-02-28 by rbrott