acmerobotics/road-runner-quickstart #233: com.acmerobotics.roadrunner imports don't work

Abraham001l, 2023-09-16:
I cloned the repositories main branch but for some reason after gradle building imports from com.acmerobotics.roadrunner don't work.

Abraham001l, 2023-09-16:
my acme robotics implementation on TeamCode is "com.acmerobotics.roadrunner:ftc:0.1.3"

rbrott, 2023-09-17:
I believe the transitive dependencies in the [pom](https://acmerobotics.z13.web.core.windows.net/com/acmerobotics/roadrunner/ftc/0.1.3/ftc-0.1.3.pom) are correct, and the project builds fine for me locally.

I suspect some combination of [invalidating caches](https://www.jetbrains.com/help/idea/invalidate-caches.html) and syncing with Gradle (Click Sync Project with Gradle Files [image]) will get Android Studio to recognize the imports.

Abraham001l, 2023-09-17:
For some reason changing to the latest version of android studio got the "com.acmerobotics.roadrunner:ftc:0.1.3" implementation working. I was peviously on version Dolphin (2021.3.1).

closed 2023-09-17 by rbrott