acmerobotics/road-runner-quickstart #238: ManualFeedFoward tuner comes up with wedged pattern vs  triangular

amjadj, 2023-09-18:
We are trying the 0.1.5 release of quickstart. Completed the push and ramp tuning and moved to Manual FeedForward. and expecting a variance with the Vref, but graph that came up was a wedged patter at 2X freq of the Vref. As in picture. What could be the issue here. 
[image]

Also, could you share  the expected range of values for kA, and also for the gains and velocity gains in Feedback tuning.

rbrott, 2023-09-19:
The problem is that your encoder velocity estimates are overflowing. I forgot to enable overflow compensation for dead wheel encoders by default. Here's the fix: https://github.com/acmerobotics/road-runner-quickstart/commit/5f2fcaf51fdab85131c71451c9b0b4544c4f20cb

EddieDL, 2023-09-19:
I was a bit confused by this for a second as well while we were upgrading. After investigating the code that @rbrott calls out. We just called 'per.setDirection(REVERSED)'. We are using dead wheels but they are connected to the ports of our drive motors. So we would have had the same issues. This works perfectly.

In your case you would just call leftFront.setDirection(REVERSED) or whatever drive encoders need to be reversed in the constructor of the DriveLocalizer. No need for extra wrappers or anything.

closed 2023-09-21 by rbrott