acmerobotics/road-runner-quickstart #242: Convert repository to template repo

ethansocal, 2023-09-21:
Would it be possible to make this repository a [template repository](https://docs.github.com/en/repositories/creating-and-managing-repositories/creating-a-template-repository)? It would avoid the issues with pull requests accidentally being submitted to this repository, make it simpler to start a new repository, and make more sense, in my opinion.

rbrott, 2023-11-29:
I did some experiments with template repositories, and they unfortunately don't quite work the way I want. Creating a repository based on a template doesn't give you the history of the template but a single squashed commit. This makes history-based upgrades difficult, and I don't want to impede that process. I would prefer a one-way fork where changes can go from upstream to downstream but not the other way around. Hopefully watchers of the repository aren't too bothered by accidental PRs and understand the tradeoff.

closed 2023-11-29 by rbrott

zachwaffle4, 2025-05-27:
Since 2023, template repositories have been updated to allow the commit history of the repository to remain, and also to allow new commits to be pushed to the repository normally. Forking the repository to allow for PRs will still be possible and recommended, but repos created from the template will not be able to do that.