acmerobotics/road-runner-quickstart #243: AngularRampLogger JS error: <number> !== <different number>

rivques, 2023-09-21:
Hi!

We're trying to set up roadrunner quickstart 1.0.0beta (cloned from the master branch of this repo). We're using a mecanum drive and drive encoders. We've gotten as far as AngularRampLogger, but we're running into a problem with the data analysis page. Here's what it looks like:
[image]
And here's the data file (extension changed to .txt so GitHub accepts it): [angular-ramp-1695328457486.txt](https://github.com/acmerobotics/road-runner-quickstart/files/12693319/angular-ramp-1695328457486.txt)
There's no line of best fit showing, and Track Width is NaN. The console shows the following error (numbers change run to run): 
```
Uncaught (in promise) Error: 3480 !== 3484
    at zl (common-6280e1ea.js:3561:96430)
    at jd (common-6280e1ea.js:3561:100020)
```
This is minified, but I couldn't find the original source anywhere (I don't know much about FTCDashboard, so it's very possible I wan't looking in the right place.) Any ideas on why this is happening and how to fix it?

We also can't find the Ramp Regression plot mentioned in [the docs we're following](https://rr.brott.dev/docs/v1-0/tuning/#angularramplogger). Does it want us to do the ForwardRampLogger, even though we have no dead wheels?

Also, what units are trackWidthTicks in? We know the track width (from CAD), can we just use the theoretical value instead of the empirical one?

Finally, is this the correct place to be asking for help? If not, where?

rbrott, 2023-09-21:
> Any ideas on why this is happening and how to fix it?

Yes, this seems like an issue I fixed very recently. Try upgrading to the latest RR FTC library: `implementation "com.acmerobotics.roadrunner:ftc:0.1.6"`.

The data you collected is good, so you just need to load the page again (and not run the logger). This is what I see:

[image]

> We also can't find the Ramp Regression plot mentioned in [the docs we're following](https://rr.brott.dev/docs/v1-0/tuning/#angularramplogger). Does it want us to do the ForwardRampLogger, even though we have no dead wheels?

What you're doing is correct. This should also be fixed by upgrading.
 
> Also, what units are trackWidthTicks in? We know the track width (from CAD), can we just use the theoretical value instead of the empirical one?

They're measured in encoder ticks. I would just use the empirical one.
 
> Finally, is this the correct place to be asking for help? If not, where?

Yes, this is a great place to ask for help.

rivques, 2023-09-21:
Thank you so much for the quick response! I'll try to implement these in the next few days.

rivques, 2023-09-22:
It worked! Thank you so much! One other thing: Our left side motors are both reversed. Because of this, ForwardPushLogger always cancelled out to (roughly) zero until we also reversed the encoders in MecanumDrive.DriveLocalizer. Is this the correct way to fix the problem? If so, should it be mentioned in the docs?

rbrott, 2023-09-22:
> It worked! Thank you so much!

:tada:

> One other thing: Our left side motors are both reversed. Because of this, ForwardPushLogger always cancelled out to (roughly) zero until we also reversed the encoders in MecanumDrive.DriveLocalizer. Is this the correct way to fix the problem? If so, should it be mentioned in the docs?

Yes, that's what you're supposed to do. Last night I updated the docs to say

[image]

after the mecanum motor debugger button mappings.

rivques, 2023-09-22:
Sounds great. Thanks for the help!

closed 2023-09-22 by rivques