acmerobotics/road-runner-quickstart #247: Manual Feedforward tuner comes up with sharp apike at peak and feedback turner goes crazy

amjadj, 2023-09-27:
We are tuning the three deadwheel odometry and the manual feedforward tuner comes up with a sharp spike at the point of direction change. Image attached. We ignored that and moved to feedback tuner. In that case the robot moves forward to the end point and then in some cases just goes to the side, or in other cases comes back, but does not stop and keeps going back. What could be happening here?
[image]

rbrott, 2023-09-28:
> We are tuning the three deadwheel odometry and the manual feedforward tuner comes up with a sharp spike at the point of direction change. Image attached. We ignored that and moved to feedback tuner. 

Bold decision! Yep, those spikes will adversely affect performance. 

> What could be happening here?

Basically your encoders are counting enough ticks to overflow the hub's velocity estimate, and the usual corrective measures failed. Can you please attach a forward ramp logger file?

teamhazmat13201, 2023-09-28:
Here is the forward ramp logger file:
[forward-ramp-1695353138957.txt](https://github.com/acmerobotics/road-runner-quickstart/files/12754439/forward-ramp-1695353138957.txt)

rbrott, 2023-10-01:
I don't see spikes in that data: 

[image]

There is quite some wobbling on one of the encoders, though I'm not sure that fully explains the behavior in the plot. Something must be going pretty wrong to get a persistently incorrect velocity as shown in the plot. In any case, I'll need a set of times (in seconds), raw encoder positions, and raw encoder velocities to try improving the overflow correction.

There's also the option of not depending on velocity measurements at all. You can avoid all velocity gains or replace the velocity with a derivative estimate (you can see the additional noise from the "online position derivative" in the plot).

teamhazmat13201, 2023-10-01:
We checked the odometers and felt that the right odometer had more friction than the left. So switched to two wheel odometry, while we rebuild the right odometer. 

However, the same issue of spike in manual feed forward is occuring. 
![Spikes in feed forward](https://github.com/acmerobotics/road-runner-quickstart/assets/57376849/b84206d8-5b8c-40ad-9b87-2
[lateral-ramp-1696183647931.txt](https://github.com/acmerobotics/road-runner-quickstart/files/12777915/lateral-ramp-1696183647931.txt)
[forward-ramp-1696183505651.txt](https://github.com/acmerobotics/road-runner-quickstart/files/12777916/forward-ramp-1696183505651.txt)
[angular-ramp-1696184537876.txt](https://github.com/acmerobotics/road-runner-quickstart/files/12777917/angular-ramp-1696184537876.txt)
4d57220009a)

teamhazmat13201, 2023-10-02:
We just changed the parallel encoder the 2 dead wheel odometry and got the same result of spikes. Ruling out issues with the Rev odometer itself.

Going to try drive encoder odometry now.

rbrott, 2023-10-02:
Well I don't see the strange behavior in those logs. Seems like there might be something special about deceleration based on the plot which won't be replicated in the logging routines. 

I would try adding the **raw** encoder position, velocity, and read time to telemetry in the manual feedforward tuner and getting a CSV from the dash logging widget.

teamhazmat13201, 2023-10-02:
> I would try adding the **raw** encoder position, velocity, and read time to telemetry in the manual feedforward tuner and getting a CSV from the dash logging widget.

Could you help us with a patch or code update of https://github.com/acmerobotics/road-runner-ftc/blob/master/RoadRunner/src/main/java/com/acmerobotics/roadrunner/ftc/Tuning.kt  for this. We are not familiar with Kotlin and cant seem to figure out what to do for this. We will create a fork off the road-runner-ftc project and try it.

rbrott, 2023-10-06:
You can copy from `Tuning.kt` if you want, though it might be easier to make your own op mode that gives a "triangular wave" pattern of motor powers/voltages.

teamhazmat13201, 2023-10-07:
> You can copy from `Tuning.kt` if you want, though it might be easier to make your own op mode that gives a "triangular wave" pattern of motor powers/voltages.
Will try this over the weekend. 

Meantime, we switched to Drive Encoders instead of dead wheels. The unusual spikes on the Manual feed forward tuner did not happen. It was a great curve fit now.

However in the feedback tuner, the robot moved forward by DISTANCE, then moved backward by 2 times DISTANCE, a very brief stop and then kept moving backward. This is the same behavior that we had the feedback tuner with deadwheel too. 

This is the log data from the dashboard.
[$Stop$Robot$ 2023-09-05 23_28_54.csv](https://github.com/acmerobotics/road-runner-quickstart/files/12836837/Stop.Robot.2023-09-05.23_28_54.csv)

We kept trying it again, and after a point, the robot started move sideways first than forward. Kept the robot on its back and tried it. The wheels would first move forward, then back, stop briefly and then continue back, stop and then go forward, stop and then go forward, then stop and then back. This continues for a while and then the wheels start moving as expected.. forward to stop, back to stop, forward, then back. We understand that the code is .linetoX(DISTANCE).linetoX(0). How is it that this behavior could occur?

rbrott, 2023-10-08:
It's unclear to me why the robot's motion was so inconsistent on `ManualFeedbackTuner`. I've moved any dependency on `drive.pose` (https://github.com/acmerobotics/road-runner-quickstart/commit/fc9b93aab48138a43847fd3c1f3e8cb7f9bda800) which is a worthwhile change, but I'm not exactly sure what's wrong with your tuning (maybe too much `kA`?). It's easier to see that issue when you plot commanded velocity against measured velocity.

I also found an issue with the overflow correction code, and I fixed it in version `0.1.7` of `com.acmerobotics.roadrunner:ftc` if you want to try dead wheels again.

madacker, 2023-10-10:
I see the same spike issue using REV through bore encoders. I can reduce the frequency by reducing maxWheelVel. Is there any chance that inverseOverflow in Encoders.kt is busted? It's suspicious to me that it's converting to uint16 when the REV result is actually a signed int16 (i.e., the overflow happens at 32767).

rbrott, 2023-10-10:
> I see the same spike issue using REV through bore encoders. I can reduce the frequency by reducing maxWheelVel. Is there any chance that inverseOverflow in Encoders.kt is busted? It's suspicious to [...]

Did you try `0.1.7`?

> I also found an issue with the overflow correction code, and I fixed it in version `0.1.7` of `com.acmerobotics.roadrunner:ftc` if you want to try dead wheels again.

Tingxu-Max-Lan, 2023-10-10:
Hi @rbrott, we have had the same issue with Rev encoders for the last several weeks. Switching to 0.1.7 has not fixed it. I recall 0.5 having an explicit solution to this. Is it not possible to implement it in 1.0?

SuhasB1, 2023-10-11:
To fix this, in your Localizer class (3 wheel or 2 wheel) Adjust the lines where you define the encoder variables to:
par0 = new OverflowEncoder(new RawEncoder(hardwareMap.get(DcMotorEx.class, "rightFront")));
par1 = new OverflowEncoder(new RawEncoder(hardwareMap.get(DcMotorEx.class, "leftRear")));
perp = new OverflowEncoder(new RawEncoder(hardwareMap.get(DcMotorEx.class, "leftFront")));

You will also need to import the OverflowEncoder in your import statement block. 

@rbrott You might want to add this in your documentation for how to get corrected values because the revcoders create overflow problems.

madacker, 2023-10-14:
0.1.7 fixes the problem in my case. I toggled back and forth a couple of times to confirm. Thanks!

rbrott, 2023-10-15:
> we have had the same issue with Rev encoders for the last several weeks. Switching to 0.1.7 has not fixed it. I recall 0.5 having an explicit solution to this. Is it not possible to implement it in 1.0?

`com.acmerobotics.roadrunner:0.1.7` has the same overflow correction as the previous quickstart (barring other bugs). If you're still seeing issues, collect a log of time, encoder position, and raw encoder velocity and open another issue.

closed 2023-10-15 by rbrott

teamhazmat13201, 2023-10-25:
Verified that the issue is resolved.