acmerobotics/road-runner-quickstart #249: Data values for Lateral Ramp Logger not showing

spanini2, 2023-09-27:
After running the Lateral Ramp Logger and clicking the "Latest" button, a graph with no values shows up. After downloading the data, we can see clearly that there is data to be plotted. We also manually connect to the control hub and notice that the same data is there. Is there a fix to this?

rbrott, 2023-09-28:
Try updating to the latest version of the library.

spanini2, 2023-10-01:
still not showing using updated quickstart, any other ideas of the source of the problem?

rbrott, 2023-10-01:
I would look for messages in the web console / network tab.

You can also try using the Android Studio file explorer to look for the files under `/sdcard/RoadRunner`.

TheCoolGuy123, 2023-10-08:
Hi! I'm from the same team.
The graph is still unable to show values (we are using the latest version that just got updated)
We even tried uploading the file from the sd card.
Here is a video: [https://drive.google.com/file/d/1rIAjVGbI25IoVWdzFvi-0mwn1dCytR8E/view?usp=sharing](url)
Any solutions?

MartianBinonwangan, 2023-10-11:
Had the same problem here. Two-Wheel Odom meccanum setup. 

Saw nothing on the graph. Downloaded and opened file to see that there was stuff to be plotted.

Have you guys been plugging in values at the top "Parameters" bar?
Nothing will show on the graph if you don't.

closed 2023-10-30 by rbrott