acmerobotics/road-runner-quickstart #253: LocalizationTest outputting Nan for robot position

jamesgu888, 2023-10-01:
Renamed encoder names to the DCmotor they were attached to
[image]

Running LocalizationTest shows NaN values in telemetry. Any idea why it isn't receiving the encoder values?
[image]

rbrott, 2023-10-01:
Have you set the encoder positions? That's (unfortunately) expected with the default encoder positions.

SuhasB1, 2023-10-11:
I have the same problem. What do you mean by "setting the encoder positions." Do you mind elaborating on this.

jamesgu888, 2023-10-11:
what rbrott said fixed my robot at least. check the threedeadwheellocalizer if you are using that and you should have to configure the variables par0YTicks, par1YTicks, and perpXTicks. 
[image]
configure these variables by running AngularRampLogger (when u go to the site there should be multiple graphs, one for trackwidthlength and other 3 for the encoder positions)

closed 2023-10-30 by rbrott