acmerobotics/road-runner-quickstart #258: Can't use FTC Dashboard to tune kS, kV and kA

madacker, 2023-10-10:
Using FTC Dashboard with ManualFeedforwardTuner and changing kS, kV and kA results in no apparent real-time changes in behavior.  Changing the values in the code and re-building _does_ change the behavior as expected, however.

Unless I'm missing something, MecanumDrive's public @Config kS, kV and kA values are copied by MotorFeedForward at constructor time and then the MecanumDrive versions are never referred to again. 

P.S. It would be great if the save-icon and the refresh-icon in FTC Dashboard had mouseover tooltips to describe what they do...

rbrott, 2023-10-10:
> Using FTC Dashboard with ManualFeedforwardTuner and changing kS, kV and kA results in no apparent real-time changes in behavior. Changing the values in the code and re-building _does_ change the [...]

Good point, fixed in https://github.com/acmerobotics/road-runner-quickstart/commit/1b1197db0cfff0e152d3c7271e9ada331333c98b.

> P.S. It would be great if the save-icon and the refresh-icon in FTC Dashboard had mouseover tooltips to describe what they do...

Opened https://github.com/acmerobotics/ftc-dashboard/issues/143

madacker, 2023-10-10:
Perhaps I'm missing something when I look at the fix but doesn't TuningOpModes.java reference MecanumDrive.feedforward to invoke the DriveView constructor so also needs to be changed?

madacker, 2023-10-10:
Never mind that last comment, I see today's commit to address the drive view factory.

madacker, 2023-10-14:
I tested the fix today and found that I still can't do real-time updates to kV, kS and kA when running ManualFeedforwardTuner, presumably because TuningOpModes again copies those parameters when constructing DriveView.

commit 42246e02 (2023-10-15): Drive view use feedforward factory (fixes #258)
closed 2023-10-15 by rbrott