acmerobotics/road-runner-quickstart #262: Removed Vector2d Rotate Capabilities

MartianBinonwangan, 2023-10-12:
I'm trying to implement field oriented drive using the 0.5.x samples and the new RoadRunner Beta. I've got everything mostly constructed, however, unlike the original sample, I can't rotate the vector2d with the robot's heading because the new class doesn't support it.

For anyone wondering, here is the original sample code from the 0.5.x quickstart guide with the area I'm getting stuck at marked:

// Read pose
Pose2d poseEstimate = drive.getPoseEstimate();

// Create a vector from the gamepad x/y inputs
// Then, rotate that vector by the inverse of that heading
Vector2d input = new Vector2d(
        -gamepad1.left_stick_y,
        -gamepad1.left_stick_x
).rotated(-poseEstimate.getHeading());  <----- PROBLEM IS HERE!

// Pass in the rotated input + right stick value for rotation
// Rotation is not part of the rotated input thus must be passed in separately
drive.setWeightedDrivePower(
        new Pose2d(
                input.getX(),
                input.getY(),
                -gamepad1.right_stick_x
        )
);

MartianBinonwangan, 2023-10-12:
Should also add that I'm not sure how to get the robot's heading. I know there is an imaginary and real component, but I don't know what to do with either.

rbrott, 2023-10-14:
See https://github.com/acmerobotics/road-runner/issues/90 and https://github.com/acmerobotics/road-runner-quickstart/issues/246

closed 2023-10-14 by rbrott