acmerobotics/road-runner-quickstart #263: Error when running ManualFeedbackTuner

SuhasB1, 2023-10-12:
When running this file, after 2 seconds of no movement, the program stops and the telemetry outputs:
java.lang.IllegalArgumentException: Path tangent orthogonal to the x-axis, try using lineToX() instead

How would we fix this error? When adjusting the function to use lineToX() rather than the lineToY(), it results in the error: Path tangent orthogonal to the x-axis, try using **lineToY()** instead. 

If easier you can also contact me on discord (suhasb)

rbrott, 2023-10-14:
What version of the library are you using?

SuhasB1, 2023-10-14:
1.0

rbrott, 2023-10-15:
1.0 isn't precise enough. I recommend upgrading `com.acmerobotics.roadrunner:ftc` to `0.1.7`.

SuhasB1, 2023-10-15:
Sorry, I mistook your question. I thought you were asking if I was using the newer version. I am using the 0.1.7 version

Kasaar, 2023-10-15:
What is your result when you run LocalizationTest? Does telemetry give you real values for your position?

SuhasB1, 2023-10-15:
Nope. I added the values from AngularRampLogger. @rbrott maybe highlighting the AngularRampLogger section to make it more clear to get the encoder values. For others using a 2 wheel setup the problem most likely comes from not changing their IMU from the default orientation to the orientation of the IMU on their actual bot. You might want to add a step to the docs.

closed 2023-10-30 by rbrott