acmerobotics/road-runner-quickstart #264: java.lang.IllegalArgumentException: Path tangent orthogonal to the x-axis, try using lineToY() instead

dewsocks, 2023-10-15:
We are going through the tuning steps on the roadrunner website and we hit the error in the title when attempting to do the ManualFeedbackTuner and are stuck. We tried changing the heading to 90 and linetoX to linetoY, neither of which worked. Changing lineToX returned the same error but in reverse. We have our kA value set to zero, and in case it changes anything our IMU is mounted upside down.

rbrott, 2023-10-15:
> We are going through the tuning steps on the roadrunner website and we hit the error in the title when attempting to do the ManualFeedbackTuner and are stuck. We tried changing the heading to 90 and [...]

What version of `com.acmerobotics.roadrunner:ftc` are you using? I recommend upgrading to `0.1.7`.

rbrott, 2023-10-15:
And I'm closing this a dupe of https://github.com/acmerobotics/road-runner-quickstart/issues/263

closed 2023-10-15 by rbrott