acmerobotics/road-runner-quickstart #266: Drive Encoder "V" Graph for Drive Encoder Angular Ramp

Monjhall, 2023-10-16:
Using drive encoders, we are receiving the following graphs for angular ramp and track width.

[image]

[image]

We saw similar behavior in #234, but we're still experiencing this issue using drive motor encoders on angular ramp. This seems to result in an invalid KV and KS, which then cause issue in the next stage of tuning. For that next stage we calculated trackWidthTicks manually, taking our measured track width of 15 inches and multiplying it by our previously determined inPerTicks value.

We've verified all motor and encoder wires are going to the correct ports in our expansion hub, that the correct motors are reversed, and that our inPerTicks and lateralInPerTicks values are close to what we'd expect theoretically.

teamhazmat13201, 2023-10-18:
We just created a new branch and go the same issue. In our case the inPerTicks were half the value expected, and we got the V pattern on the AngularRampLogger. We had another branch where all of it is working correctly! After a lot of careful comparison, turned out that when we reversed the Encoders in DriveLocalizer() we had copied and pasted the REVERSE-ing lines from MecanumDrive(). But the encoder names are different! In MacanumDrive(), the motors are leftBack and rightBack. In DriveLocalizer(), the encoder names are leftRear and rightRear. Once we corrected it, we had correct inPerTicks and the correct ramp pattern in AngularRampLogger.

Hope this helps.

rbrott, 2023-10-18:
Having both rear and back was silly. I've replaced uses of rear with back.

closed 2023-10-30 by rbrott