acmerobotics/road-runner-quickstart #270: My robot is breaking when I run Manual Feedback Tuner and Spline Test

rockit456, 2023-10-21:
When I run both of these tests, the wheels will move forward for a second then start going back and forth so fast. The robot starts shaking and doesn't really move from it's original position. Has anyone ever had this problem?

leonkun007, 2023-10-22:
same issue here. I did not see any issue for previous steps.  
With the latest PR 1.0x .

rockit456, 2023-10-25:
yes the issue is weird

teamhazmat13201, 2023-10-25:
We had the same issue today while calibrating with 3 dead wheel encoders. We had completed all the way till Angular Ramp Logger. And when we started the Manual Feed Forward Tuner, the robot would just vibrate on the spot. At this time we had not downloaded the code with the updated values. We were just using the dashboard update the values.
The issue went away, when we downloaded the code with the new values and started Manual Feed Forward Tuner.

rockit456, 2023-10-25:
oh ok this is very helpful!

Zw96042, 2023-10-26:
We are having the same issue. We have tried inputting the values into the code and it is still just vibrating on the spot. If anyone has any other solutions that would be very helpful.

rbrott, 2023-10-26:
Any information you can share (library versions, drive parameters, dashboard plots, logs, etc.) would be helpful.

Zw96042, 2023-10-26:
The drive parameters we are using are:
public double inPerTick = 0.003;
public double lateralInPerTick = 0.002;
public double trackWidthTicks = 5318.3;

// feedforward parameters (in tick units)
public double kS = 0.67;
public double kV = 0.0005;
public double kA = 0.00015;

// path profile parameters (in inches)
public double maxWheelVel = 50;
public double minProfileAccel = -30;
public double maxProfileAccel = 50;

// turn profile parameters (in radians)
public double maxAngVel = Math.PI; 
public double maxAngAccel = Math.PI;

// path controller gains
public double axialGain = 0.0;
public double lateralGain = 0.0;
public double headingGain = 0.0; 

public double axialVelGain = 0.0;
public double lateralVelGain = 0.0;
public double headingVelGain = 0.0; 

My dependencies for the TeamCode module are:
implementation project(':FtcRobotController')
annotationProcessor files('lib/OpModeAnnotationProcessor.jar')

implementation "com.acmerobotics.roadrunner:ftc:0.1.8"
implementation "com.acmerobotics.dashboard:dashboard:0.4.13"
implementation 'com.acmerobotics.roadrunner:core:1.0.0-beta3'
implementation 'com.acmerobotics.roadrunner:actions:1.0.0-beta3'
implementation 'org.ftclib.ftclib:core:2.0.1'

implementation 'com.acmerobotics.roadrunner:core:0.5.6'

My dependencies for the root project are:
implementation 'org.firstinspires.ftc:gameAssets-CenterStage:1.0.0'
implementation 'org.tensorflow:tensorflow-lite-task-vision:0.4.3'
runtimeOnly 'org.tensorflow:tensorflow-lite:2.12.0'
implementation 'androidx.appcompat:appcompat:1.2.0'
implementation 'com.acmerobotics.dashboard:dashboard:0.4.12'

I do not have any errors in the console and I will check to see what the graph shows when I get a chance.

Zw96042, 2023-10-27:
Our team has changed it to a two dead wheel localizer and even though we have 2 dead wheels it seems to be working for now. Can we make sure that the 3 dead wheel localizer is working correctly?

rbrott, 2023-10-30:
TLDR: Save your dead wheel and other params! The values from the regressions must be set in code (they do not automatically take effect).

Pretty sure I know what's going on now (I finally got my hands on a log!). RR 1.0 currently uses the estimated/actual pose instead of the target pose to compute the body-frame feedforward. This gives a better feedforward at the cost of making it dependent on localization. I'd be interested to hear thoughts on that tradeoff.

So why does that make the robot go crazy? Well the log that I have has an incredible angular velocity:
[image]
Why is the localizer so wrong? Well it turns out the parameters are left at their defaults:
[image]
Super small track width => small left/right encoder differences balloon into massive rotations. I originally had all zeros to start but that gave some annoying `NaN`s after the zero division. Not sure if that's better or worse -- I should probably just fail the op mode if the defaults are still in place.

> Our team has changed it to a two dead wheel localizer and even though we have 2 dead wheels it seems to be working for now. Can we make sure that the 3 dead wheel localizer is working correctly?

The 2 dead wheel localizer "works" with the default parameters because you don't need to do anything special to get a reasonable angular velocity (even if you're using pitch/roll, the values at least won't be insane).

closed 2023-10-30 by rbrott