acmerobotics/road-runner-quickstart #272: How do you perform a custom action in the middle of a path?

EddieDL, 2023-10-25:
We have the desire to have one seamless path that also includes things like "Move Arm", "Score". In the past we have customized the Trajectory Following logic to accept a function that would be run every loop. In that function we would check where the robot is and if it is in a specific range we perform some action.

I think we could do something similar using Custom Actions in a Parallel action? But we would like to know if that would be a good or recommended approach.

EddieDL, 2023-10-25:
One other note, I do not believe we would want to just chain trajectory actions with custom actions. The reason is that we do not want to have to come to a complete stop to perform our custom actions.

In the FRC world we use Path Planner which has the concept of Markers which can have Actions. Something like that would be ideal, but I don't think that is available here.

rbrott, 2023-10-26:
> We have the desire to have one seamless path that also includes things like "Move Arm", "Score". In the past we have customized the Trajectory Following logic to accept a function that would be run [...]

That same strategy works fine here if you call `run()` each loop on an `Action` instead of `Actions.runBlocking()`.

> I think we could do something similar using Custom Actions in a Parallel action? But we would like to know if that would be a good or recommended approach.

If you like the actions framework, you can do exactly what you suggested: write & compose custom actions and run them alongside the trajectory following action.

> One other note, I do not believe we would want to just chain trajectory actions with custom actions. The reason is that we do not want to have to come to a complete stop to perform our custom actions.

Right, you would perform your custom action and the trajectory following action in parallel, not in sequence.

> In the FRC world we use Path Planner which has the concept of Markers which can have Actions. Something like that would be ideal, but I don't think that is available here.

The marker concept no longer exists in RR, though you can accomplish the same thing with `afterDisp()` and `afterTime()` in `TrajectoryActionBuilder`.

closed 2023-10-26 by EddieDL