acmerobotics/road-runner-quickstart #273: Having trouble uploading the data to RR Lateral Ramp Regression

rockit456, 2023-10-26:
I have pressed latest multiple times and reran the code multiple times but no data is showing up on the graph. Does anyone have any ideas to fix this?

Zw96042, 2023-10-26:
I believe you have to input the values that it is asking for above the latest button.

rockit456, 2023-10-26:
I have inputed kS, kV, and inchesPerTick

rbrott, 2023-10-26:
Can you [open the console](https://developer.chrome.com/docs/devtools/open/#shortcuts) and report any messages?

Also what version of the libraries are you running?

raghuponugoti, 2023-10-26:
We had the same problem. In the documentation it says for (mecanum dead wheels only) so was wasnt sure if it is for Three dead wheel Odometry as well?
Didnt had issues to any other Logger. 
We are using 0.1.8

rockit456, 2023-10-27:
No errors stood out in the console and we are using the latest version of the quickstart

rockit456, 2023-10-27:
This was working a couple weeks ago in an older version

closed 2023-11-26 by rbrott