acmerobotics/road-runner-quickstart #275: How should we think about SetTangent

EddieDL, 2023-10-26:
How should we think about when to use SetTangent? What is the use case for when to use it?

EddieDL, 2023-11-01:
I think the only other question I have is how to think about where the tangent starts or is defaulted to.

rbrott, 2023-11-02:
Some builders expose the initial tangent as a parameter. For example, [here's `PositionPathSeqBuilder`](https://github.com/acmerobotics/road-runner/blob/9c2a6287a6a07177983ec1d226a14db63b9ad6dc/core/src/main/kotlin/com/acmerobotics/roadrunner/Builders.kt#L26).

But the higher-level ones set the initial tangent based on the initial heading. If you want to change that, you can call `setTangent()` immediately before any other methods.

EddieDL, 2023-11-02:
If you call `setTangent(90).lineToX(10)`, what will be the tangent be if you call `lineToY()`? Does the tangent reset every trajectory?

When you try to set the tangent is it relative to the current heading?

We think we know answers to these, but it is hard to visualize all the time.

rbrott, 2023-11-03:
> If you call `setTangent(90).lineToX(10)`, what will be the tangent be if you call `lineToY()`?

Not quite sure what you mean here. `setTangent(Math.toRadians(90)).lineToX(10)` won't work because the tangent is pointing orthogonal to the x-axis.

> Does the tangent reset every trajectory?

No, the tangent by default is the end tangent of the last path/trajectory segment.

> When you try to set the tangent is it relative to the current heading?

No, the tangent is relative to the x-axis / global coordinate system. It's decoupled totally from the heading.

closed 2023-11-11 by rbrott

EddieDL, 2023-11-13:
I am not sure that the tangent is 100% right. I think that might only apply to the `setTangent` method. I think when you add a tangent to one of the spline methods it is based on the current orientation of the robot... or something. Is that right?

rbrott, 2023-11-13:
> I am not sure that the tangent is 100% right. I think that might only apply to the `setTangent` method. 

Not really sure what you mean here.

> I think when you add a tangent to one of the spline methods it is based on the current orientation of the robot... or something. Is that right?

Do you have an example where tangent/heading interact in an unexpected way? Here's an example where changing heading doesn't affect the spline path:
* Before: https://rr.brott.dev/playground/?94c2dd079e6c711d
* After: https://rr.brott.dev/playground/?f6f413099d3619d4