acmerobotics/road-runner-quickstart #279: Awesome visualizer!

EddieDL, 2023-11-01:
I am not sure if it is documented anywhere but the visualizer used in this https://github.com/acmerobotics/road-runner-quickstart/issues/275#issuecomment-1784410145 response is pretty awesome. I wish you could impose the field over it :) but it is awesome and people should use it.

I know it isn't part of this library, but I am not sure where I might be able to ask about it. I am wondering if you could use this to visualize vectors? We are having a hard time visualizing what something like `mecdrive.pose.heading.inverse().times(new Vector2d(input.x, input.y));` does. 

We are trying to use this code to do field relative driving, which sort of works. Our problem is that as you rotate the robot the heading used for driving changes. So if you turn 90 degrees right is left and left is right or something similar. I was hoping to be able to visualize the vector so we could better understand what the multiplication does since multiplying vectors isn't familiar to us.

rbrott, 2023-11-02:
> I was hoping to be able to visualize the vector so we could better understand what the multiplication does since multiplying vectors isn't familiar to us.

The key is to imagine the rotation is more like [a matrix](https://en.wikipedia.org/wiki/Rotation_matrix) than another vector. The operation `rot.times(vec)` is applying `rot` to `vec`, rotating `vec` by the rotation represented by `rot`. 

> Our problem is that as you rotate the robot the heading used for driving changes. So if you turn 90 degrees right is left and left is right or something similar. 

Left and right completely swapping sounds more like a 180deg rotation than a 90deg one. If the rotation is wrong, try removing the `inverse()`. You can also try running small programs/using the REPL to experiment with the API:
```
val r = Rotation2d.exp(Math.toRadians(90.0))
 val x = Vector2d(1.0, 0.0)
 r.times(x)
res9: com.acmerobotics.roadrunner.Vector2d = Vector2d(x=6.123233995736766E-17, y=1.0)
```

EddieDL, 2023-11-02:
This helps. The internal values of Rotation2d are called real and imag which we did some testing to figure out those are basically the sin and cos for the heading. 

Are these the two values used in the matrix? Meaning the matrix is something like [sin, cos] that represents the angle. We are thinking we can just graph those values as X and Y to see the angle.

I am not sure what REPL is. It seems like you are just doing the operation and then putting the new vector to the output. The big challenge is that requires the robot. Which is why that visualizer is so great, we can have someone on the team working on the path without needing access to the robot or any of the hardware.

We did notice shortly after we posted this the ability to edit things in the playground and play them went away. Sorry if this was meant to be hidden. I am sure you don't need another tool to maintain. It is just such a great way to figure out paths and we have not seen anything similar for FTC. Is there anything we could do to help make that tool available or is there the possibility to take a version of it and run it locally?

rbrott, 2023-11-03:
> This helps. 

> The internal values of Rotation2d are called real and imag which we did some testing to figure out those are basically the sin and cos for the heading.

Yes, `(real, imag)` corresponds with `(cos, sin)`. (You can also think of a rotation as a unit complex number.)

> Are these the two values used in the matrix? Meaning the matrix is something like [sin, cos] that represents the angle. We are thinking we can just graph those values as X and Y to see the angle.

Yep, those two values can be used to build a rotation matrix. If you want to get the angle itself, call `log()`.

> I am not sure what REPL is. 

A REPL is a [read–eval–print loop](https://en.wikipedia.org/wiki/Read%E2%80%93eval%E2%80%93print_loop). Basically an interactive prompt for entering in program expressions. I use the [IntelliJ Kotlin REPL](https://www.jetbrains.com/help/idea/kotlin-repl.html) inside a `road-runner` project for my experimentation.

> It seems like you are just doing the operation and then putting the new vector to the output. The big challenge is that requires the robot. Which is why that visualizer is so great, we can have someone on the team working on the path without needing access to the robot or any of the hardware.

You don't need a robot to do these experiments. You should be able to make a class with a `main()` method in any IntelliJ project and run the code with the green arrow that appears without deploying to the robot.

> We did notice shortly after we posted this the ability to edit things in the playground and play them went away. Sorry if this was meant to be hidden. 

No worries, this is meant to be used.

> is there the possibility to take a version of it and run it locally?

It's not quite ready to run locally, though I may release it in some form in the future.

EddieDL, 2023-11-13:
We got things sorted out. Thanks for the help. Looking forward to when the visualizer is ready.

closed 2023-11-13 by EddieDL