acmerobotics/road-runner-quickstart #286: ManualFeedbackTuner error

TheMujdii3, 2023-11-09:
When running manual feedback tuner it gives me this:
[image]

rh0dium, 2023-11-09:
What does line 59 of org.firstinspires.ftc.teamcode.OurCode.M_Auto.runOplode(M_Auto.java:59) Look like?

rbrott, 2023-11-10:
I believe the requirement is https://github.com/acmerobotics/road-runner/blob/9c2a6287a6a07177983ec1d226a14db63b9ad6dc/core/src/main/kotlin/com/acmerobotics/roadrunner/Profiles.kt#L349:
```kotlin
    require(maxVels.all { v -> v > 0.0 })
```
Presumably you set the maximum velocity constraint to a bad value.

TheMujdii3, 2023-11-10:
> What does line 59 of org.firstinspires.ftc.teamcode.OurCode.M_Auto.runOplode(M_Auto.java:59) Look like?
> 
> 

runBlocking(drive.actionbuilder(pose2d)....build) something like that

TheMujdii3, 2023-11-10:
> I believe the requirement is https://github.com/acmerobotics/road-runner/blob/9c2a6287a6a07177983ec1d226a14db63b9ad6dc/core/src/main/kotlin/com/acmerobotics/roadrunner/Profiles.kt#L349: [...]

i ll try on monday