acmerobotics/road-runner-quickstart #288: ManualFeedbackTuner - Not working at all?

rh0dium, 2023-11-09:
Gobilda Mecanum Drive System
Gobilda Odemetry Pods

Ref:  https://discord.com/channels/225450307654647808/1165782776327442472/1165782776327442472
Ref: https://discord.com/channels/225450307654647808/1164076500010668033
Ref: https://discord.com/channels/225450307654647808/1155945941912785009

```
        public double inPerTick = 0.002938133725342;  // Defaults to 0
        public double lateralInPerTick = -0.0019094444; // Defaults to 1
        public double trackWidthTicks =  3915.1594498081;  // Defaults to 0

        public double kS = 1.1218644019; // Defaults to 0
        public double kV =  0.0004388153001; // Defaults to 0
        public double kA = 0.00007; // Defaults to 0

```

We've managed to get overthing working up till this Step.   ManualFeedbackTuner is not doing anything - it appears we aren't alone.  Are we supposed to see the bot go back and forth?  Is there a graph to tune like ManualFeedforwardTuner?

Are there loose ranges for these numbers?   What can we do to help you?

rh0dium, 2023-11-10:
So we made some progress - Turns out the IMU was pulling NaN values.  Ok fixed that.  Now it's jerking around and having a small seizure.  No values seem to help.

rbrott, 2023-11-10:
> Now it's jerking around and having a small seizure.

Sounds like #270

closed 2023-11-26 by rbrott