acmerobotics/road-runner-quickstart #297: end() for recording consecutive sequences?

madacker, 2023-11-16:
For recording multiple TrajectoryActionSequences that are to be run consecutively, I don't see any way to retrieve the end-pose of a sequence. On 0.5.x, TrajectorySequence::end() could be used for this purpose. Is there a new way to do this?

rbrott, 2023-11-17:
My original idea was for teams to just have one big `TrajectoryActionBuilder` and create functions to chunk things up. But I'm open to suggestions to replace `end()` in the old system. Writing an `end()` for a generic `Action` is annoying and won't always produce a result. Perhaps a variant of `build()` could return the end pose and the final action as a pair.

madacker, 2023-11-18:
Yeah, one big TrajectoryActionBuilder isn't always feasible. An example is when the robot has to traverse some amount to take a sensor reading and use that to choose one of a number of subsequent potential trajectories. The choice can't be pre-baked. Each of those potential trajectories will have the same start-pose as the end-pose of the prior trajectory. 

Maybe just a "getCurrentPose()" hanging off TrajectoryActionBuilder, if you don't mind messing up the builder pattern (so no patterns like builder.getCurrentPose().build())? From what I've seen, teams can't do one long builder pattern this year anyway and are breaking into pieces for modularity of handling blue/red, near/far, left/middle/right-prop, etc.

rbrott, 2023-11-18:
Making [`lastPoseUnmapped`](https://github.com/acmerobotics/road-runner/blob/9c2a6287a6a07177983ec1d226a14db63b9ad6dc/actions/src/main/kotlin/com/acmerobotics/roadrunner/Actions.kt#L155) public is definitely an option (and probably less confusing than returning an unmapped pose from a method that ends a (mapped) trajectory). 

Another possibility is a builder method that instantiates a new builder with the same pose + tangent + constants as the old one. Then you could do something like
```kotlin
val sharedStartBuilder = TrajectoryActionBuilder(...).splineTo(...)
val sharedStart = sharedStartBuilder.build()
val continuation1 = sharedStartBuilder.fresh().lineToY(...).build()
val continuation2 = sharedStartBuilder.fresh().splineTo(...).build()

Actions.runBlocking(sharedStart)
if (cond) {
    Actions.runBlocking(continuation1)
} else {
    Actions.runBlocking(continuation2)
}
```

rbrott, 2023-11-26:
I added `fresh()`, and I'm curious if it handles your use cases.

25wangj, 2023-12-23:
Perhaps a `SwitchAction` would work in this situation - something that takes in a lambda that evaluates to a condition and a `Map` of `Action`s corresponding to the potential results.

zachwaffle4, 2024-11-11:
I know this hasn't been touched in almost a year at this point, but would it be possible to create a `TrajectorySequenceAction` class that is returned by `.build()` and has an `endPose` property? This would reduce confusion around `.fresh()` (because there is apparently a lot of that) and stop people from needing separate variables for the unbuilt and built trajectory sequences.

rbrott, 2024-11-11:
> This would reduce confusion around `.fresh()` (because there is apparently a lot of that)

I can see how `fresh()` is different from the old `end()` that teams are familiar with, though I still think `fresh()` has superior semantics. I continue to think that `endPose` leads to bugs when used with pose mapping, and it doesn't capture the last tangent. One useful property of any TAB-splitting functionality is that the two halves can be joined into one TAB without changing the shape of the path.

I hope the confusion around `fresh()` is mostly due to some (regrettable) broken behavior (https://github.com/acmerobotics/road-runner/issues/111) and not with its overall design.

> and stop people from needing separate variables for the unbuilt and built trajectory sequences.

I don't really see the need for separate variables. Users should get in the habit of storing the final TAB in a variable and then doing `tab.build()` wherever the `Action` is needed. (This also prevents the common bug of reusing an `Action` -- one can `build()` as many `Actions` from the same `tab` as they would like.)

zachwaffle4, 2024-11-11:
I imagine that most of the issues lie in the lack of documentation for `.fresh()` other than this issue (and now acmerobotics/road-runner#111). Adding a page about trajectory splitting to your docs page would probably be a good idea. 

> I hope the confusion around fresh() is mostly due to some (regrettable) broken behavior and not with its overall design.
I think this is true, yes. In my experience on the FTC Discord people arent using `.endTrajectory()` before `.fresh()` so that would contribute to the overall confusion.

> I continue to think that `endPose` leads to bugs when used with pose mapping, and it doesn't capture the last tangent.
What kind of bugs do you think `endPose` would lead to? I also don't see why having a `TrajectorySequenceAction` with an `endPose` would need current TAB-splitting to change, it would only add new features to what happens to a TAB *after* it's built.

rbrott, 2024-11-12:
> What kind of bugs do you think `endPose` would lead to?

With pose mapping, there are two choices for `endPose`. Each has disadvantages. The mapped pose makes the most sense in that it matches the actual end of the trajectory if one were to query it, but it's not the right pose to pass into the next TAB. The unmapped pose has the opposite problem. Either choice comes with some unintuitive behavior and hence bugs.

> I also don't see why having a `TrajectorySequenceAction` with an `endPose` would need current TAB-splitting to change, it would only add new features to what happens to a TAB _after_ it's built.

It might help to illustrate my point with an example. The autonomous here
```kotlin
val traj1 = actionBuilder(Pose2d(0.0, 0.0, 0.0)).setTangent(PI).splineTo(Vector2d(10.0, -10.0), -PI / 2)
val traj2 = traj1.fresh().splineTo(Vector2d(20.0, 0.0), 0)
runBlocking(traj1.build())
runBlocking(traj2.build())
```
has the same behavior as the obvious combination
```kotlin
val traj = actionBuilder(Pose2d(0.0, 0.0, 0.0)).setTangent(PI).splineTo(Vector2d(10.0, -10.0), -PI / 2)
  .endTrajectory().splineTo(Vector2d(20.0, 0.0), 0)
runBlocking(traj.build())
```
but the seemingly similar `endPose` variation has different behavior:
```kotlin
val traj1 = actionBuilder(Pose2d(0.0, 0.0, 0.0)).setTangent(PI).splineTo(Vector2d(10.0, -10.0), -PI / 2)
val traj2 = actionBuilder(traj1.build().endPose).splineTo(Vector2d(20.0, 0.0), 0)
runBlocking(traj1.build())
runBlocking(traj2.build())
```

To fix this, the tangent needs to be propagated as well, adding to the noise.

zachwaffle4, 2024-11-13:
>  Either choice comes with some unintuitive behavior and hence bugs.

If that's the case, having separate properties `endPoseMapped` and `endPoseUnmapped` would help users understand how they're supposed to use both.

> but the seemingly similar endPose variation has different behavior:

My proposal would still allow people to use `.fresh()` from unbuilt trajectories, again allowing for both possibilities. I think the following is actually more intuitive, at least based on the way I have been using RoadRunner:

```kt
val traj1 = actionBuilder(Pose2d(0.0, 0.0, 0.0)).setTangent(PI).splineTo(Vector2d(10.0, -10.0), -PI/2).build()
val traj2 = actionBuilder(traj1.endPose).splineTo(Vector2d(20.0, 0.0), 0).build()

runBlocking(SequentialAction(traj1, traj2))
```

> To fix this, the tangent needs to be propagated as well, adding to the noise.

I don't understand what you mean by propagation here.

rbrott, 2024-11-17:
> I don't understand what you mean by propagation here.

Sorry, I just mean that `actionBuilder(traj1.endPose)` isn't enough. You really need `actionBuilder(traj1.endPose).setTangent(traj1.endTangent)` in the general case.

closed 2025-01-12 by rbrott