acmerobotics/road-runner-quickstart #300: Actions for TeleOp?

madacker, 2023-11-18:
Actions are great! But they might be even greater if the same Actions could be used for concurrency in TeleOp as well as Auton. (Maybe they are, but I didn't spot a way.) A couple of potential ways pop to mind:

1. Let an Action generically spawn another concurrent Action. Then the main loop itself could be converted to an Action.
2. Keep the main loop as not-an-Action but let it add to an Action queue, and let it invoke Action processing. (Technically people could create this solution on their own today to allow come Actions between Auton and TeleOp, but it might be a more compelling case for Actions if they could work in both modes out of the box.)

rbrott, 2023-11-18:
Option 2) appeals to me. The teleop routine can maintain a list of ongoing actions. In each iteration, the op mode calls `run()` on every action, removing the ones that return `false`. This is very possible with the current actions, though I haven't mentioned the pattern anywhere.

rbrott, 2023-11-19:
I left drawing out of my earlier comment. This is a good place to start: https://github.com/acmerobotics/road-runner-ftc/blob/389a7c9d50915a6637a5d395e74a4948b4a37ed0/RoadRunner/src/main/java/com/acmerobotics/roadrunner/ftc/Actions.kt#L13

rbrott, 2023-11-26:
I wrote a short guide on this: https://rr.brott.dev/docs/v1-0/guides/teleop-actions/

closed 2023-11-26 by rbrott