acmerobotics/road-runner-quickstart #303: RoadRunner Manual Feedback Tuner Issue (1.0.x Beta)

cougar15, 2023-11-19:
Hello, I am in team 19890.
We are using version 1.0.x Beta of RoadRunner and are running into a problem when we are using the Manual Feedback Tuner. We are using mecanum drive with three dead wheels. Everything was running smoothly before this and the localization test works perfectly.
For some reason, the robot is going forward and then starts strafing left and back. We are not sure why this is happening, and have tried changing the different gains to fix the problem, bit this did not work. Others we have asked have only used the last version, but we would prefer using the new version. We greatly appreciate any help! Thank you.
-Team 19890

rbrott, 2023-11-20:
Can you please retrieve robot logs? (see http://localhost:1313/docs/v1-0/tuning/#getting-help)

cougar15, 2023-11-20:
Hello,
Thanks for replying. The robot logs are attached and so is the robot movement video. Thank you!
- Team 19890
[Robot Movement 11-20-2023 Manual Feed Back Tuner.zip](https://github.com/acmerobotics/road-runner-quickstart/files/13419417/Robot.Movement.11-20-2023.Manual.Feed.Back.Tuner.zip)

[2023_11_20__15_51_15_630__ManualFeedbackTuner.log](https://github.com/acmerobotics/road-runner-quickstart/files/13419420/2023_11_20__15_51_15_630__ManualFeedbackTuner.log)

rbrott, 2023-11-21:
The estimated heading is clearly vary much more than it should: 
[image]
Hard to know whether it's the cause or just a symptom (another reason to consider removing any effect of estimated pose on feedforward computation). I assume the the large movements in the middle are from user intervention, and the odd stabilization close to 90deg is affected by some wheel slip. Does the situation improve when you start increasing `headingGain`?

cougar15, 2023-11-21:
Hello,
I didn't intervene with the robot while it was moving, but I did notice the wheels slipping a bit. I will test increasing headingGain and will let you know what happens. Thank you!
-Team 19890

AwesomePerson01, 2023-11-21:
Hi!
We increased the headingGain and the robot started turning instead of strafing back and forth.  
The logs are attached below
Thanks!

- Team 19890

[2023_11_21__16_07_42_958__ManualFeedbackTuner.log](https://github.com/acmerobotics/road-runner-quickstart/files/13430901/2023_11_21__16_07_42_958__ManualFeedbackTuner.log)

rbrott, 2023-11-23:
If you didn't intervene and the heading spike like that, something is seriously wrong. Is it possible that someone swapped the encoders on the bot or something like that? I would run `AngularRampLogger` again and see if the wheel positions have changed.

AwesomePerson01, 2023-11-23:
All of the deadwheel positions stayed basically the same, except for par0, which changed from 1048.246580934229 to 33.76369003638886.  Do you know why this is happening?  
 Thanks,
 - Team 19890

rbrott, 2023-11-23:
Does that result match the IRL configuration (that is, the first parallel wheel is much closer to the center of the robot than the second parallel wheel)? Probably worth making sure that the encoders don't need to be swapped.

cougar15, 2023-11-23:
When we tried retuning, we got values for the localization that seemed right. However now, the robot seems to move randomly when we run the manual feed back tuner.
We swapped the encoders sometime ago, and that helped then, but it isn't doing anything now. When we do increase the heading gain, it doesn't help. What should we do now? Thank you for any help!
- Team 19890

AwesomePerson01, 2023-11-24:
Hi, we tried running the angular ramp logger, and everything seems to be normal ex
[image]
cept for par0.  This is the regression graph we got for it.  Should this be happening?

cougar15, 2023-11-24:
Hello, 
We have figured out the problem and have fixed it. The dead wheel was slipping, but we remounted it. Thank you for all your help!

closed 2023-11-24 by rbrott