acmerobotics/road-runner-quickstart #307: Autonomy model or smth...

TheMujdii3, 2023-11-21:
Hello there!Finally our code works well and we are in the phase where we cant make a model for our autonomies.Can someone help us with a model or some basic autonomy?Pls

rbrott, 2023-11-23:
I'm not in the best position to help there. Consider talking to the folks in the [FTC DIscord server](https://discord.com/invite/first-tech-challenge). Once you have a specific goal in mind, I can help you use the RR API to design and coordinate various motions.

closed 2023-11-23 by rbrott

TheMujdii3, 2023-11-23:
yeah i kinda have problems with linetoy but ...

rbrott, 2023-11-23:
I can probably help if you post the code

TheMujdii3, 2023-11-23:
yeah here
package org.firstinspires.ftc.teamcode.OurCode;

import com.acmerobotics.roadrunner.Action;
import com.acmerobotics.roadrunner.Pose2d;
import com.acmerobotics.roadrunner.Vector2d;
import com.acmerobotics.roadrunner.ftc.Actions;
import com.qualcomm.robotcore.eventloop.opmode.Autonomous;

import com.qualcomm.robotcore.eventloop.opmode.LinearOpMode;

import org.firstinspires.ftc.teamcode.MecanumDrive;
import org.opencv.core.Mat;

@Autonomous
public class Blue_Short extends LinearOpMode {
    Robothardware robot = new Robothardware();

    @Override
    public void runOpMode()  {
        robot.init(hardwareMap);

        MecanumDrive drive  = new MecanumDrive(hardwareMap,new Pose2d(-12,-58.26,Math.toRadians(0)));

        Action drive_to_drop = drive.actionBuilder(drive.pose)
                //.lineToY(-56)
                //.lineToXLinearHeading(-56, Math.toRadians(90))
                //.lineToYSplineHeading()
                .lineToY(3)
                
                .splineTo(new Vector2d(-12., -50), Math.toRadians(90))
                .endTrajectory()
                .build();
        waitForStart();
        Actions.runBlocking(drive_to_drop);
        drive.updatePoseEstimate();

    }
}

this the first part of my auto

rbrott, 2023-11-23:
Ah. The start pose has a heading of 0deg, and the action builder will assume that the starting tangent is 0deg as well. If you want to move sideways, you need to change the tangent to 90deg before `lineToY()`. See https://rr.brott.dev/playground/?cbc25a58b9280d8b.

TheMujdii3, 2023-11-23:
Ohhhh thank u soooo much