acmerobotics/road-runner-quickstart #308: Spline Test Do Not Achieve The Expectation.

lovezfy, 2023-11-23:
### RR FTC Version

0.1.8

### Observed Behavior

We are using version 1.0.x Beta of RoadRunner and are running into a problem. We are using mecanum drive with (mecanum drive encoders only), when we are using the "splineTo(new Vector2d(24, 24), Math.PI / 2)", the robot turned 180 degrees, not 90 degrees as expected.  I can't understand this phenomenon. Please, ask for help.

https://github.com/acmerobotics/road-runner-quickstart/assets/32976854/dcf05e5f-b917-438c-969b-773fa5bba0d8

### Robot Logs

[2023_11_23__10_30_19_055__SplineTest.log](https://github.com/acmerobotics/road-runner-quickstart/files/13445637/2023_11_23__10_30_19_055__SplineTest.log)

rbrott, 2023-11-23:
Localization in the log seems to match IRL, so I think the encoders are configured + calibrated correctly. The motion also looks roughly correct which mostly rules out swapped/reversed motors. That leaves the constraints -- can you try cutting all of those parameters (`maxAngVel`, maxAngAccel`, etc.) in half and running again? I plan to also log the command in the future which would be helpful here.

rbrott, 2023-11-25:
```
MECANUM_PARAMS
  axialGain: 0.0
  axialVelGain: 0.0
  headingGain: 0.0
  headingVelGain: 0.0
  inPerTick: 0.03352
  kA: 0.00015
  kS: 1.5387481614909282
  kV: 0.0043076533352661645
  lateralGain: 0.0
  lateralInPerTick: 0.0355
  lateralVelGain: 0.0
  maxAngAccel: 3.141592653589793
  maxAngVel: 3.141592653589793
  maxProfileAccel: 50.0
  maxWheelVel: 50.0
  minProfileAccel: -30.0
  trackWidthTicks: 776.0

Channel                Messages         Hz
ESTIMATED_POSE              133      59.64
TARGET_POSE                 133      59.84
```

It's also weird to me that `lateralInPerTick` is greater than `inPerTick`...

closed 2025-01-12 by rbrott