acmerobotics/road-runner-quickstart #310: Cancelable Trajectory

Zw96042, 2023-11-24:
I don't see any method or type anywhere on the docs about canceling trajectories even though it has that in New Features. Am I missing something? Could I have help with the syntax for canceling trajectories?

rbrott, 2023-11-24:
I'll preface this by distinguishing abrupt cancellation from smooth cancellation. Abrupt cancellation entails immediately stopping the current action and commanding a new one. Teams may be familiar with this from [LRR's interrupting a live trajectory section](https://learnroadrunner.com/advanced.html#interrupting-a-live-trajectory). Smooth cancellation involves coming to a stop as soon as possible while still respecting the constraints of the trajectory. Cancellation as mentioned in "new features" and elsewhere in RR core refers to smooth cancellation, and the rest of this comment outlines how to achieve that.

Cancellation is supported by the core library through the `Trajectory` type (**not** `CancelableTrajectory` as the docs still say...), though it isn't exposed by `TrajectoryActionBuilder` or in the quickstart. (I may extend `TrajectoryActionBuilder` to support cancellation natively by producing a `CancellableAction` or something under the hood.)

To get a `Trajectory`, you can use `TrajectoryBuilder` (`build()` gives you a sequence of them). You can call `cancel()` on that object to get a `DisplacementTrajectory` that eases the robot to a stop. One can imagine putting this together into an `Action` or something that can switch from following the main (base) trajectory and failing over to a cancellation trajectory when commanded. 

In general, the foundations are there for a sufficiently ambitious team. It's just not quite ready and mature enough for everyone else.

Zw96042, 2023-11-24:
Thank you for your response. I am not sure what other parameters to feed into TrajectoryBuilder. I am just giving it my drive position and it is requesting others which I do not know what they mean. Could you briefly explain the parameters needed for the TrajectoryBuilder? On LRR it just shows it taking the Pose2D. Could you also explain the offset parameter in the cancel method? This is what I currently have but it is requesting more parameters.

```
Trajectory toRedWing = new TrajectoryBuilder(drive.pose)
                    .splineToConstantHeading(new Vector2d(0.00, 11.00), Math.toRadians(180.00))
                    .splineToConstantHeading(new Vector2d(-32.00, 11.00), Math.toRadians(180.00))
                    .splineToLinearHeading(new Pose2d(-53.00, 54.00, Math.toRadians(-45.00)), Math.toRadians(180.00))
                    .build();
```

rbrott, 2023-11-25:
> Thank you for your response. I am not sure what other parameters to feed into TrajectoryBuilder. I am just giving it my drive position and it is requesting others which I do not know what they mean. Could you briefly explain the parameters needed for the TrajectoryBuilder?

See [the docs](https://rr.brott.dev/docs/v1-0-0-beta1/core/kdoc/core/com.acmerobotics.roadrunner/-trajectory-builder/). These defaults are generally reasonable: https://github.com/acmerobotics/road-runner-quickstart/blob/df17f93fd3e037d58332ce5ac29f732d670c5881/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/MecanumDrive.java#L424-L431.

> On LRR it just shows it taking the Pose2D. 

LRR is not updated for 1.0. The official 1.0 docs and code are you best references.

> Could you also explain the offset parameter in the cancel method?

I'm not sure what you mean by the offset parameter. `Trajectory#cancel()` takes one parameter that specifies from where in the original trajectory you want to come to a stop.

Zw96042, 2023-11-25:
I have looked at those default values, and I am a bit confused still about the input parameters. Looking at those defaults I can see that is for the TrajectoryActionBuilder which I believe I cannot use for the cancel() method. If this is not the case then please let me know. 

If there is an example you could provide me with that would be great because I still do not understand things like eps, resolution, and posemap. 

Is that cancel parameter required? I was looking at the declaration and to me I believe it said something about offset but will double check when I am able to. The parameter name was 's'. I am looking to cancel it at the closest possible point to the robot but still want to use this smooth stop instead of an abrupt stop.

If you can please help me further understand this that would be great.

rbrott, 2023-11-25:
> I have looked at those default values, and I am a bit confused still about the input parameters. Looking at those defaults I can see that is for the TrajectoryActionBuilder which I believe I cannot use for the cancel() method. If this is not the case then please let me know.

Correct, you can't use `cancel()` with TAB, but it shares many parameters with `TrajectoryBuilder`.

> If there is an example you could provide me with that would be great because I still do not understand things like eps, resolution, and posemap.

Set `eps` to `1e-6` (controls tolerance for arclength parameterization) and `resolution` to `0.25` (controls displacement sampling of constraints for profile generation).

> Is that cancel parameter required? I was looking at the declaration and to me I believe it said something about offset but will double check when I am able to. The parameter name was 's'. I am [...]

Yes, the parameter to `cancel()` is required. The `Trajectory` class doesn't have any knowledge of what the "closest possible point to the robot is." The higher-level system that follows trajectories needs to convey that notion to `Trajectory` by reducing it to a displacement along the trajectory (along the way, you may need to go from elapsed time to displacement along the trajectory).

Zw96042, 2023-11-25:
Thank you very much for your guidance on this matter. That clears up the values for me. This might be a silly question but how would I declare a variable as a Trajectory type and have the path like this? I believe I would have to declare this variable but am not sure what this includes. The docs seem to currently be down at the moment so I am unable to check those for reference.

`Actions.runBlocking(
                    new TrajectoryBuilder(drive.pose, 1e-6, 0.0, drive.defaultVelConstraint, drive.defaultAccelConstraint, 0.25, 0.1)
                        .splineToConstantHeading(new Vector2d(0.00, 11.00), Math.toRadians(180.00))
                        .splineToConstantHeading(new Vector2d(-32.00, 11.00), Math.toRadians(180.00))
                        .splineToLinearHeading(new Pose2d(-53.00, 54.00, Math.toRadians(-45.00)), Math.toRadians(180.00))
                        .build());`

rbrott, 2023-11-26:
> This might be a silly question but how would I declare a variable as a Trajectory type and have the path like this? I believe I would have to declare this variable but am not sure what this includes. 

`TrajectoryBuilder` builds a `Trajectory` which is not an `Action`. You'll need to first turn it into a `TimeTrajectory` with `new TimeTrajectory(traj)` and then create a `FollowTrajectoryAction`.

closed 2023-12-03 by Zw96042