acmerobotics/road-runner-quickstart #311: How do we use Actions without Init?

marvelsofmas, 2023-11-24:
Hey,

I am trying to follow along with the [action tab](https://rr.brott.dev/docs/v1-0/actions/). I am fairly lost. How am I supposed to run motor/servo commands by calling the action class without **init**?

This is the example given:
[image]

This is what we have:
Robot Class:
[image]

Auton Class:
[image]

robot.groundWrist()....will never do anything. If I remove the method the class itself loses reference to the motor/servo commands. Thank you.

marvelsofmas, 2023-11-24:
Nevermind, we figured it out. But seriously an absurd amount of boiler plate code to run what was originally dead simple with Trajectory sequences in 0.5.X.

closed 2023-11-24 by marvelsofmas

rbrott, 2023-11-24:
> Hey,
> 
> I am trying to follow along with the [action tab](https://rr.brott.dev/docs/v1-0/actions/). I am fairly lost.

Totally fair. That page is out of date, and I'm working to update it soon. 

> How am I supposed to run motor/servo commands by calling the action class without **init**?

As the error on the `@Override` annotation suggests, `init()` is no longer part of the `Action`. You should instead move all of that into `run()` along with `return false`. You can even shorten that into a lambda since `Action` is a SAM interface.

> Nevermind, we figured it out. But seriously an absurd amount of boiler plate code to run what was originally dead simple with Trajectory sequences in 0.5.X.

Sorry for the headache. In the future, feel free to post here earlier on. I'm happy to field any questions about RR 1.0. Hopefully the lambda tip reduces the boilerplate compared to trajectory sequences.