acmerobotics/road-runner-quickstart #320: AngularRampLogger doesn't automatically graph it's results

ftc19853, 2023-11-30:
We recently used the AngularRampLogger and it was working with no problem, spinning around and "updating" the graph when the button was pressed. We updated to 1.0.0Beta4 and now it doesn't show anything on the graph.
[image]
It also shows that whenever we actually import any file as well.

rbrott, 2023-11-30:
I don't believe the update should have changed anything. Can you please download the tuning file and attach it here?

ftc19853, 2023-12-01:
I don't think this is the latest one(we switch pcs some time:)
[angular-ramp-1699736019208.json](https://github.com/acmerobotics/road-runner-quickstart/files/13519730/angular-ramp-1699736019208.json)

rbrott, 2023-12-01:
Can you report any errors in the web console? It all looks fine to me locally:
[image]

ftc19853, 2023-12-01:
is there supposed to be a clump at the end?
[image]

rbrott, 2023-12-02:
Yes, if you leave the logger running after it finishes the ramp then it will accumulate samples there.

closed 2023-12-02 by rbrott