acmerobotics/road-runner-quickstart #323: Plotty.js pages (forward-ramp, lateral-ramp, dead-wheel-angular-ramp etc.) include and exclude buttons don't work as expected

ToddInIowa, 2023-12-05:
### RR FTC Version

0.1.9

### Observed Behavior

When a box is drawn on the plot and you click the exclude button the whole graph looks to be excluded, the slope does change slightly.  When I press the include button nothing changes, even though the slope should change.
[image]
Screenshot of an attempt to change the slope aggressively via include.

[image]
Screenshot trying to exclude some plots.

Chrome version 119.0.6045.160
Edge version 119.0.2151.97

I didn't see any error in the Chrome debugging console.

### Tuning Files

[lateral-ramp-1701752252825.json](https://github.com/acmerobotics/road-runner-quickstart/files/13555254/lateral-ramp-1701752252825.json)
[angular-ramp-1701753355254.json](https://github.com/acmerobotics/road-runner-quickstart/files/13555255/angular-ramp-1701753355254.json)
[forward-ramp-1701752149737.json](https://github.com/acmerobotics/road-runner-quickstart/files/13555256/forward-ramp-1701752149737.json)

rbrott, 2023-12-05:
The plotly interface is confusing. You won't see the final selection after including/excluding until you exit the box selection tool. I usually exit by double clicking on the plot. Here's a demo I recorded that hopefully shows how it works:

https://github.com/acmerobotics/road-runner-quickstart/assets/6677314/bfe57348-9c82-49c1-b211-7fc62b5bc3c2

rbrott, 2023-12-17:
I see what you mean about the controls though. I just released a new version of the tuning tools that downgrades Plotly to restore the old (desired) select behavior.

closed 2023-12-17 by rbrott