acmerobotics/road-runner-quickstart #327: OpMode Initialization Not Reseting Pose Properly

Monjhall, 2023-12-09:
### RR FTC Version

0.1.9

### Observed Behavior

When trying to tune our feedback parameters we noticed the robot was not driving the proper distance forward and then drove a significantly increased distance backwards. Looking at the FTC dashboard telemetry we noticed our X value was starting very large, causing the robot to complete its forward path early. Then when driving backwards it had a much greater distance to cover.

To verify a pose reset wasn't occurring on init we used the LocalizerTest opmode to drive around, stop, and then check what our initial position appeared to be. This made the problem appear even stranger, with the robot seeming to "start" from an average of the previous X + Y, not 0.

Note: We are running on carpet instead of the field tiles due to an issue with our IMU failing. Common sentiment online is that this is caused by ESD. We've ordered some Staticide, but haven't received it yet to apply to our field.

Video of the Behavior:
https://youtu.be/dC9c_xn_zd0

### Robot Logs

[2023_12_08__21_32_50_482__LocalizationTest.log](https://github.com/acmerobotics/road-runner-quickstart/files/13622437/2023_12_08__21_32_50_482__LocalizationTest.log)

rbrott, 2023-12-09:
Can you post your code? I suspect the issue is caused by reading some drive encoders before reversing them and then reading them after. The TODO comments in the quickstart were misleading until I just pushed https://github.com/acmerobotics/road-runner-quickstart/commit/02d8ea76154635bf773a0465e226ca9e41c16fd5.

Monjhall, 2023-12-09:
I can't post the code until Tuesday unfortunately, since I'm out of state. I can confidently say that we reversed one motor under the motor TODO specifying as such and the same encoder where the encoder TODO specified.

Looking at that commit, I can see what you're saying about the potential issues. We can pull the latest on Tuesday and get back to you on if it fixed the issue.

Thanks for looking into this so promptly!

Monjhall, 2023-12-09:
(Closed by accident)

Monjhall, 2023-12-13:
I was able to pull the changes and this looks great. Hopefully we will be able to tune feedback parameters this week and start running spline test! Thanks for the help!

(Closed on purpose)

closed 2023-12-13 by Monjhall